myriadrf/LimeSuite #130: Wrong transfer data rate on first run

darkstar007, 2017-06-24:
I'm currently running the latest git version (17.06). The first time I run a capture (after I've done a rm -rf ~/.limesuite) to do a 4,000,000S/s, single channel, complex int16 capture it says:

'''
[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 9060A02593414'
[INFO] Estimated reference clock 30.7196 MHz
[INFO] Selected reference clock 30.720 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] Init LMS7002M(0)
[INFO] creating directory /home/matt/.limesuite
[INFO] LMS7002M cache /home/matt/.limesuite/LMS7002M_cache_values.db
[INFO] Ver=7, Rev=1, Mask=1
[INFO] LMS7002M calibration values caching Enable
CGEN: Freq=80 MHz, VCO=2.56 GHz, INT=82, FRAC=349525, DIV_OUTCH_CGEN=15
M=156, N=3, Fvco=1040.000 MHz
16: FF AF AA
16: FF AF 3C
[snip]
M=131, N=1, Fvco=1048.000 MHz
Sampling at 4000000 S/s on chan 0.
Tuned to 432000000 Hz on channel 0.
****Antenna set to LNAL on chan 0
MCU algorithm time: 1 ms
MCU Ref. clock: 30.72 MHz
MCU algorithm time: 156 ms
Bandwidth parameter 3500000 Hz resulted in 3500000 Hz on chan 0.
Tuner gain set to 90.00 dB on chan 0.
Reading samples in sync mode...
############################################################
Rx calibration using RSSI INTERNAL ON BOARD loopback
Rx ch.A @  432 MHz, BW: 3.5 MHz, RF input: LNAL, PGA: 31, LNA: 15, TIA: 3
Performed by: MCU
------------------------------------------------------------
MCU algorithm time: 1 ms
Current MCU firmware: 3, DC/IQ calibration full
MCU Ref. clock: 30.72 MHz
MCU algorithm time: 198 ms
Rx: 3.205 MB/s
Rx: 3.202 MB/s
Rx: 3.202 MB/s
Rx: 3.202 MB/s
Rx: 3.202 MB/s
'''

So its doing a 3.2MB/s transfer, to transfer 4Ms/s * sizeof(complex int16)!

The second time I run the command (with no removal of the ~/.limesuite dir) I get:
'''
[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 9060A02593414'
[INFO] Estimated reference clock 30.7196 MHz
[INFO] Selected reference clock 30.720 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] Init LMS7002M(0)
[INFO] LMS7002M cache /home/matt/.limesuite/LMS7002M_cache_values.db
[INFO] Ver=7, Rev=1, Mask=1
[INFO] LMS7002M calibration values caching Enable
CGEN: Freq=80 MHz, VCO=2.56 GHz, INT=82, FRAC=349525, DIV_OUTCH_CGEN=15
M=156, N=3, Fvco=1040.000 MHz
16: 01 A0 AA
16: AA AA 2A
[snip]
Sampling at 4000000 S/s on chan 0.
SetFrequency using cache values vco:2, csw:119
Tuned to 432000000 Hz on channel 0.
****Antenna set to LNAL on chan 0
[INFO] Rx Filter calibrated from cache
Bandwidth parameter 3500000 Hz resulted in 3500000 Hz on chan 0.
Tuner gain set to 90.00 dB on chan 0.
Reading samples in sync mode...
############################################################
Rx calibration using RSSI INTERNAL ON BOARD loopback
Rx ch.A @  432 MHz, BW: 3.5 MHz, RF input: LNAL, PGA: 31, LNA: 15, TIA: 3
Rx calibration: using cached values
Rx calibration values found in cache:
   | DC  | GAIN | PHASE
---+-----+------+------
I: |   0 | 2047 | 0
Q: |   0 | 2047 |
############################################################
[INFO] L 1044480
Rx: 16.073 MB/s
Rx: 16.073 MB/s
'''

What appears to the correct data rate (and the data rate given appears to match the actual data rate) is actually given this time. However, looking at the data recorded, I'm not convinced its correct - but I'll look further into this.

I've tried to use SopaySDRDevice_getChannelInfo() to confirm the version of the driver I am using but it returns nothing.

Thanks,

Matt

jocover, 2017-06-29:
I have the same issue and full duplex mode not working

IgnasJarusevicius, 2017-06-30:
I haven't looked at MCU code but I have checked this by running calibration from LimeSuiteGUI and it restores registers after calibration. The streaming gets messes up if you run calibration while streaming is active (fftViewer running) though.

guruofquality, 2017-07-03:
The calibrations are occurring after `IConnection::SetupStream` and before `IConnection::ControlStream`. This corresponds to the hooks in SoapySDR for setupStream and activateStream. I dont know if calling CalibrateRx/Tx should be an issue after SetupStream, even though the stream has not yet been started with ControlStream. If thats is something we should fix or not?

If we have to fix it, basically I am looking for a good place to run the MCU calibrations before the device is actually streaming. Many applications actually setup the stream before applying settings, so there arent a lot of options here to maintain flexibility. -- I was also thinking of delaying the first call to `_conn->SetupStream(streamID, config)` by moving it to inside of activateStream() instead. Its not ideal, but that would less us schedule all stream setup on the limesdr until after the calibration code is run.

pseudocode:
```
setupStream()
{
    //save requested channel setup in StreamConfig
    StreamConfig config;
    ....
}

activateStream()
{
    if (dir == SOAPY_SDR_RX) getRFIC(ch)->CalibrateRx(_actualBw.at(dir).at(ch));
    if (dir == SOAPY_SDR_TX) getRFIC(ch)->CalibrateTx(_actualBw.at(dir).at(ch));
    _conn->SetupStream(streamID, config);
    _conn->ControlStream(i, true);
}
```

Thoughts on that as a solution?