myriadrf/LimeSuite #132: connectivity issue with Lime Suite GUI

usmanfarooq948, 2017-06-29:
I am working on GUI of Lime Suite. I am facing a an issue of connectivity of Lime Suite GUI Application with LMS7002m 
EVB Ver 2 hardware.

Old version of Lime Suite i:e Version: 16.3.804 is successfully detected the LMS7002 hardware device. As shown in diagram (s) below.

[image]
[image]
also showing ports in connection window (as shown in image below).

[image]

But in latest Lime Suite GUI Application version. It is unable to connect. Even not showing ports. As shown in below image.

[image]

Kindly purpose me a solution of above mentioned issue.

IgnasJarusevicius, 2017-06-29:
Binary builds on GitHub are now longer updated.
If you are using older chip revision (LMS7002mr2), try:
http://downloads.myriadrf.org/builds/limesuite/LimeSuiteGUI_MR2_478db355.exe
or if you have recent board with new chip ((LMS7002mr3):
http://downloads.myriadrf.org/builds/limesuite/LimeSuiteGUI_20170616_6524d2.exe
If you are compiling from source, COM connection is not enabled by default, you need to enable it in CMake (set ENABLE_EVB7COM to true).

usmanfarooq948, 2017-06-30:
Thanks for helping sir, Kindly provide me source of Lime Suite for LMS7002mr3.

IgnasJarusevicius, 2017-06-30:
You can get source code for Lime Suite from master (latest) or stable (more testing done) branch. If you want exact commit for the binary version linked above, its filename contains part of git commit hash (use 'git checkout 6524d2')

usmanfarooq948, 2017-07-03:
Respected sir please give me the guide(including examples of RX and TX) of LimeSuiteGUI_20170616_6524d2.exe

IgnasJarusevicius, 2018-06-22:
Closing old issue.

closed 2018-06-22 by IgnasJarusevicius