myriadrf/LimeSuite #136: cmake build outputs to the wrong directory.

romeojulietthotel, 2017-08-29:
I used to do this

```
cd LimeSuite/src/
mkdir build
cd build
cmake ../
make
```

But that doesn't work because cmake outputs to the wrong directory. Hmm. So then I try this:

```
cd LimeSuite
mkdir build
cd build
cmake ../
```

But that too outputs into LimeSuite.  The generally accepted way to use cmake is to use

```
cd LimeSuite
mkdir build
cd build
cmake ../
make
```

9600, 2017-08-29:
Please see:

http://wiki.myriadrf.org/Lime_Suite#Building_LimeSuite

closed 2017-08-29 by 9600

romeojulietthotel, 2017-08-29:
I had always built from `LimeSuite/src/build `and I ran into these issues.  It turns out that even when I tried to build from `LimeSuite/build` the cmake output was going into the top-level dir.

I finally ran `git clean -di ` and then ran cmake again from `LimeSuite/build` and got a clean cmake and make.

But since `LimeSuite/build` has source files  it's still messy mixing source with build only files.  `LimeSuite/build` should be an empty directory so that it's easy to do a clean cmake and make.

So to be clear `LimeSuite/build/bin` and `LimeSuite/build/mcu_program` should be moved  to the top level.