myriadrf/LimeSuite #137: MacOS compilation / brew fails at `lms7_api.cpp`

ralfbiedert, 2017-09-20:
Running `brew install limesuite` on MacOS produces:

```
/tmp/limesuite-20170920-98654-mojyyb/src/API/lms7_api.cpp:738:14: error: ordered comparison between pointer and zero ('float_type *' (aka 'double *') and 'int')
    if (gain < 0)
        ~~~~ ^ ~
/tmp/limesuite-20170920-98654-mojyyb/src/API/lms7_api.cpp:760:14: error: ordered comparison between pointer and zero ('unsigned int *' and 'int')
    if (gain < 0)
        ~~~~ ^ ~
2 errors generated.

```

IgnasJarusevicius, 2017-09-20:
Its definitely an error. It should be (*gain < 0). I will fix this asap.
I don't have MacOS to test on though.

ralfbiedert, 2017-09-21:
Compiles just fine now with `brew install limesuite --HEAD`. 

We probably still need a new release ([v17.09.0.tar.gz](https://github.com/myriadrf/LimeSuite/archive/v17.09.0.tar.gz) contains the same problem) to fix regular `brew`.

Thanks!

IgnasJarusevicius, 2017-09-22:
I have created release v17.09.1 tag that includes this fix. Closing the issue.

closed 2017-09-22 by IgnasJarusevicius