myriadrf/LimeSuite #142: "clock_locked" always returns false in SoapyLMS7

ccsh23, 2017-10-04:
Dear all,

I have found following bug: No matter what master clock rate I am trying to set,"clock_locked" sensor always returns false when trying to read its value with SoapySDR API, though I have checked that in theory it is implemented in SoapyLMS7 plugin. I don't think error is in my code, as I can successfully read "lo_locked" sensors for TX and RX PLLs, which basically uses the same method and type of returned value.

guruofquality, 2017-10-04:
hey @ccsh23 I'm a one liner python kind of guy. :-) The following prints true for me:

* `python -c "import SoapySDR; d = SoapySDR.Device('driver=lime'); d.setMasterClockRate(80e6); print(d.readSensor('clock_locked'))"`

I'm curious how it runs for you, or if you have a value that is misbehaving here.

ccsh23, 2017-10-04:
Oh Gosh, I was using the same name for string and boolean variables holding the sensor value and that was the reason of a problem in my code. Sorry for taking your time and thank you for a quick response, thanks to it I was able to see this :)

closed 2017-11-11 by guruofquality