myriadrf/LimeSuite #145: FFT Viewer Causes segfault with Wayland

btashton, 2017-10-16:
When using Wayland which is becoming the default on many Linux distributions a segfault is triggered when FFT Viewer is loaded
The root of this issue is this:
http://trac.wxwidgets.org/ticket/17702

Forcing X11 backend works around this issue: `GDK_BACKEND=x11 ./bin/LimeSuiteGUI`

Traceback can be seen here

```
Thread 1 "LimeSuiteGUI" received signal SIGSEGV, Segmentation fault.
0x00007ffff21720ed in _XSend () from /lib64/libX11.so.6
Missing separate debuginfos, use: dnf debuginfo-install GConf2-3.2.6-17.fc26.x86_64 SDL2-2.0.5-4.fc26.x86_64 at-spi2-atk-2.24.1-1.fc26.x86_64 at-spi2-core-2.24.1-1.fc26.x86_64 atk-2.24.0-1.fc26.x86_64 bzip2-libs-1.0.6-22.fc26.x86_64 cairo-1.14.10-1.fc26.x86_64 cairo-gobject-1.14.10-1.fc26.x86_64 dbus-glib-0.108-2.fc26.x86_64 dbus-libs-1.11.16-1.fc26.x86_64 dconf-0.26.0-2.fc26.x86_64 expat-2.2.4-1.fc26.x86_64 fontconfig-2.12.5-1.fc26.x86_64 freetype-2.7.1-9.fc26.x86_64 gdk-pixbuf2-2.36.9-1.fc26.x86_64 glib2-2.52.3-1.fc26.x86_64 graphite2-1.3.10-1.fc26.x86_64 gtk3-3.22.21-1.fc26.x86_64 gvfs-client-1.32.1-2.fc26.x86_64 harfbuzz-1.4.4-1.fc26.x86_64 jbigkit-libs-2.1-6.fc26.x86_64 libICE-1.0.9-9.fc26.x86_64 libSM-1.2.2-5.fc26.x86_64 libX11-1.6.5-2.fc26.x86_64 libXau-1.0.8-7.fc26.x86_64 libXcomposite-0.4.4-9.fc26.x86_64 libXcursor-1.1.14-8.fc26.x86_64 libXdamage-1.1.4-9.fc26.x86_64 libXext-1.3.3-5.fc26.x86_64 libXfixes-5.0.3-2.fc26.x86_64 libXi-1.7.9-2.fc26.x86_64 libXinerama-1.1.3-7.fc26.x86_64 libXrandr-1.5.1-2.fc26.x86_64 libXrender-0.9.10-2.fc26.x86_64 libXxf86vm-1.1.4-4.fc26.x86_64 libblkid-2.30.2-1.fc26.x86_64 libcap-2.25-5.fc26.x86_64 libcroco-0.6.12-1.fc26.x86_64 libdatrie-0.2.9-4.fc26.x86_64 libepoxy-1.4.3-1.fc26.x86_64 libffi-3.1-12.fc26.x86_64 libgcc-7.2.1-2.fc26.x86_64 libgcrypt-1.7.8-1.fc26.x86_64 libglvnd-0.2.999-24.20170818git8d4d03f.fc26.x86_64 libglvnd-egl-0.2.999-24.20170818git8d4d03f.fc26.x86_64 libglvnd-glx-0.2.999-24.20170818git8d4d03f.fc26.x86_64 libgpg-error-1.25-2.fc26.x86_64 libjpeg-turbo-1.5.1-0.fc26.x86_64 libmount-2.30.2-1.fc26.x86_64 libmspack-0.6-0.1.alpha.fc26.x86_64 libpng-1.6.28-2.fc26.x86_64 librsvg2-2.40.18-1.fc26.x86_64 libselinux-2.6-7.fc26.x86_64 libstdc++-7.2.1-2.fc26.x86_64 libthai-0.1.25-2.fc26.x86_64 libtiff-4.0.8-1.fc26.x86_64 libusbx-1.0.21-2.fc26.x86_64 libuuid-2.30.2-1.fc26.x86_64 libwayland-client-1.13.0-1.fc26.x86_64 libwayland-cursor-1.13.0-1.fc26.x86_64 libxcb-1.12-3.fc26.x86_64 libxkbcommon-0.7.1-3.fc26.x86_64 libxml2-2.9.4-2.fc26.x86_64 lz4-libs-1.8.0-1.fc26.x86_64 mesa-libwayland-egl-17.2.2-2.fc26.x86_64 pango-1.40.12-1.fc26.x86_64 pcre-8.41-1.fc26.x86_64 pixman-0.34.0-3.fc26.x86_64 sqlite-libs-3.20.1-1.fc26.x86_64 systemd-libs-233-6.fc26.x86_64 wxBase3-3.0.3-5.fc26.x86_64 wxGTK3-3.0.3-5.fc26.x86_64 wxGTK3-gl-3.0.3-5.fc26.x86_64 xz-libs-5.2.3-2.fc26.x86_64 zlib-1.2.11-2.fc26.x86_64
(gdb) where
#0  0x00007ffff21720ed in _XSend () at /lib64/libX11.so.6
#1  0x00007ffff21687d4 in XQueryExtension () at /lib64/libX11.so.6
#2  0x00007ffff0aa8052 in __glXLookupDisplay () at /lib64/libGLX.so.0
#3  0x00007ffff0aa40d5 in glXQueryVersion () at /lib64/libGLX.so.0
#4  0x00007ffff5814be5 in wxGLCanvasX11::GetGLXVersion() () at /lib64/libwx_gtk3u_gl-3.0.so.0
#5  0x00007ffff5815255 in wxGLCanvasX11::ConvertWXAttrsToGL(int const*, int*, unsigned long) () at /lib64/libwx_gtk3u_gl-3.0.so.0
#6  0x00007ffff5815a18 in wxGLCanvasX11::InitXVisualInfo(int const*, __GLXFBConfigRec***, XVisualInfo**) () at /lib64/libwx_gtk3u_gl-3.0.so.0
#7  0x00007ffff5816c00 in wxGLCanvas::Create(wxWindow*, int, wxPoint const&, wxSize const&, long, wxString const&, int const*, wxPalette const&) () at /lib64/libwx_gtk3u_gl-3.0.so.0
#8  0x00007ffff5816da3 in wxGLCanvas::wxGLCanvas(wxWindow*, int, int const*, wxPoint const&, wxSize const&, long, wxString const&, wxPalette const&) () at /lib64/libwx_gtk3u_gl-3.0.so.0
#9  0x00000000005b0693 in OpenGLGraph::OpenGLGraph(wxWindow*, int, wxPoint const&, wxSize const&, long, wxString const&, int const*) (this=0x3fa5800, parent=0xd0f820, id=-1, pos=..., size=..., style=524288, name=..., args=0x8acde0 <OpenGLGraph::GLCanvasAttributes>) at /home/bashton/limesdr/tools/LimeSuite/src/oglGraph/OpenGLGraph.cpp:68
#10 0x000000000058aaac in frFFTviewer::frFFTviewer(wxWindow*, int, wxString const&, wxPoint const&, wxSize const&, long) (this=0x407a000, parent=0xe54400, id=-1, title=..., pos=..., size=..., style=541597248)
    at /home/bashton/limesdr/tools/LimeSuite/src/fftviewer_wxgui/fftviewer_wxgui.cpp:40
#11 0x000000000058ff16 in fftviewer_frFFTviewer::fftviewer_frFFTviewer(wxWindow*) (this=0x407a000, parent=0xe54400) at /home/bashton/limesdr/tools/LimeSuite/src/fftviewer_wxgui/fftviewer_frFFTviewer.cpp:50
#12 0x000000000056561e in LMS7SuiteAppFrame::OnShowFFTviewer(wxCommandEvent&) (this=0xe54400, event=...) at /home/bashton/limesdr/tools/LimeSuite/src/lms7suiteAppFrame.cpp:348
#13 0x00007ffff5c05c8a in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /lib64/libwx_baseu-3.0.so.0
#14 0x00007ffff5c0609a in wxEvtHandler::SearchDynamicEventTable(wxEvent&) () at /lib64/libwx_baseu-3.0.so.0
#15 0x00007ffff5c0612f in wxEvtHandler::TryHereOnly(wxEvent&) () at /lib64/libwx_baseu-3.0.so.0
#16 0x00007ffff5c061e3 in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /lib64/libwx_baseu-3.0.so.0
#17 0x00007ffff5c06245 in wxEvtHandler::ProcessEvent(wxEvent&) () at /lib64/libwx_baseu-3.0.so.0
#18 0x00007ffff62ed67b in wxWindowBase::TryAfter(wxEvent&) () at /lib64/libwx_gtk3u_core-3.0.so.0
#19 0x00007ffff5c05fa7 in wxEvtHandler::SafelyProcessEvent(wxEvent&) () at /lib64/libwx_baseu-3.0.so.0
#20 0x00007ffff62a16bb in wxMenuBase::SendEvent(int, int) () at /lib64/libwx_gtk3u_core-3.0.so.0
#21 0x00007ffff619390b in menuitem_activate () at /lib64/libwx_gtk3u_core-3.0.so.0
#22 0x00007ffff279030d in g_closure_invoke () at /lib64/libgobject-2.0.so.0
#23 0x00007ffff27a298e in signal_emit_unlocked_R () at /lib64/libgobject-2.0.so.0
#24 0x00007ffff27ab1a5 in g_signal_emit_valist () at /lib64/libgobject-2.0.so.0
#25 0x00007ffff27abb0f in g_signal_emit () at /lib64/libgobject-2.0.so.0
#26 0x00007ffff37e34f6 in gtk_widget_activate () at /lib64/libgtk-3.so.0
#27 0x00007ffff36b2d66 in gtk_menu_shell_activate_item () at /lib64/libgtk-3.so.0
#28 0x00007ffff36b3042 in gtk_menu_shell_button_release () at /lib64/libgtk-3.so.0
#29 0x00007ffff36963b7 in _gtk_marshal_BOOLEAN__BOXEDv () at /lib64/libgtk-3.so.0
#30 0x00007ffff2790546 in _g_closure_invoke_va () at /lib64/libgobject-2.0.so.0
#31 0x00007ffff27aae69 in g_signal_emit_valist () at /lib64/libgobject-2.0.so.0
#32 0x00007ffff27abb0f in g_signal_emit () at /lib64/libgobject-2.0.so.0
#33 0x00007ffff37e0f34 in gtk_widget_event_internal () at /lib64/libgtk-3.so.0
#34 0x00007ffff369328e in propagate_event () at /lib64/libgtk-3.so.0
#35 0x00007ffff36953b0 in gtk_main_do_event () at /lib64/libgtk-3.so.0
#36 0x00007ffff31a6685 in _gdk_event_emit () at /lib64/libgdk-3.so.0
#37 0x00007ffff3202b62 in gdk_event_source_dispatch () at /lib64/libgdk-3.so.0
#38 0x00007ffff24b8247 in g_main_context_dispatch () at /lib64/libglib-2.0.so.0
#39 0x00007ffff24b85e8 in g_main_context_iterate.isra () at /lib64/libglib-2.0.so.0
#40 0x00007ffff24b8902 in g_main_loop_run () at /lib64/libglib-2.0.so.0
#41 0x00007ffff36944e5 in gtk_main () at /lib64/libgtk-3.so.0
#42 0x00007ffff611c8a5 in wxGUIEventLoop::DoRun() () at /lib64/libwx_gtk3u_core-3.0.so.0
#43 0x00007ffff5ac7b43 in wxEventLoopBase::Run() () at /lib64/libwx_baseu-3.0.so.0
#44 0x00007ffff5a8fb46 in wxAppConsoleBase::MainLoop() () at /lib64/libwx_baseu-3.0.so.0
#45 0x00007ffff5b18087 in wxEntry(int&, wchar_t**) () at /lib64/libwx_baseu-3.0.so.0
#46 0x000000000057b524 in main(int, char**) (argc=1, argv=0x7fffffffdaa8) at /home/bashton/limesdr/tools/LimeSuite/src/lms7suiteApp.cpp:25
```

blitz, 2018-05-05:
I have the same issue on Fedora 28. It took a while to figure out this is a X11/Wayland issue.

+1 for proper Wayland support.

oystercatcher, 2019-02-08:
thanks  had not figured out why  seg fault with fftviewer  on  fedora 29

log out and login with   xorg        fftviewer no crash

http://nuclearrambo.com/wordpress/getting-limesdr-mini-to-work-for-the-first-time/

ztamosevicius, 2019-05-13:
This is wxWidgets related.

closed 2019-05-13 by ztamosevicius