myriadrf/LimeSuite #149: Read/Write GPIO at specific hardware time

ccsh23, 2017-11-08:
Just wanted to ask if such enhancement (reading/writing GPIO at specific hardware time, referably accessable via SoapySDR API) would be possibile? It would be very helpful e.g. when implementing half duplex systems - I know at which hardware time I want to transmit and receive, but would like to set GPIO controlling RF switch at these moments.

tblake84, 2017-11-09:
I’m seconding this. Need a simple way to run half-duplex with a T/R switch.

guruofquality, 2017-11-09:
This is a related post on the subject: https://discourse.myriadrf.org/t/harmonic-operation/849/5

Rather than any sort of time based switching, the idea is to use the tx sample stream activity in the FPGA to switch between two register states for the GPIO. I think this methodology may be simpler to implement, although it still needs FPGA support.

9600, 2017-11-09:
Tagging @ztamosevicius for comment on possibility of FPGA implementation.

ccsh23, 2017-11-09:
@guruofquality, if I got it right you are talking about something very similar to GPIO API supported by some USRP models (see link below)?
https://files.ettus.com/manual/page_gpio_api.html

That would be perfect for half-duplex-based applications based on LimeSDR!

IgnasJarusevicius, 2017-12-18:
I think the latest LimeSDR FPGA gateware has an ability to toggle GPIO pin based on TX activity:
https://github.com/myriadrf/LimeSDR-USB_GW/commit/81652db61f9f0cd44579137645482c173dad3e1f

Currently, there is no dedicated API for changing GPIO pin functions. However, it probably just needs a write to some register to change pin function. So It would be possible do enable it manually using register write function. I will find out and post the details tomorrow. Maybe, we could enable it by default in SoapyLMS.

ccsh23, 2017-12-19:
Great news, I'll be looking forward to it :)

IgnasJarusevicius, 2017-12-19:
GPIO_0 (J19 header pin 1) function can be changed by 0x00C0[0] bit in FPGA registers. In short, writing 0xFFFE to FPGA register 0x00C0 enables Tx activity indication on GPIO_0. I think Soapy function writeRegister(0x00C0, 0xFFFE) should do the trick.

When the feature is enabled the GPIO_0 goes high, when samples are being sent to LMS chip.
To test it, you need GW 2.13:
https://github.com/myriadrf/LimeSDR-USB_GW/raw/master/output_files/LimeSDR-USB_lms7_trx_HW_1.4.rbf

laf0rge, 2019-08-08:
I think this feature is very useful way beyond a trigger at TX activation.  In any TDMA system (GSM, TETRA, LTE), you would normally e.g. want to output a frame pulse every time the transmitter starts to transmit the first burst within a frame.  This is then used to synchronize various test equipment to the start of the frame, so you can do measurements in sync with the frame clock.

Also, in burst-oriented transmitter schemes, you might want to ramp up a PA before the burst and ramp it down after the burst is over, with defined timing/offset to the actual burst you're sending.

The way how I would enviison to do this (and I believe it has been done with USRP before) is that you can insert a "timed command" in the stream to the FPGA which then would set/unset a GPIO at a given specified sample clock.