myriadrf/LimeSuite #151: LimeSuite fails to build on Debian / armv7l and aarch64

laf0rge, 2017-11-29:
At Osmocom, we're having trouble building LimeSuite for armv7l and aarch64.

See e.g. https://build.opensuse.org/package/live_build_log/network:osmocom:nightly/limesuite/Debian_9.0/armv7l
or https://build.opensuse.org/package/live_build_log/network:osmocom:nightly/limesuite/Debian_9.0/aarch64

In both cases the problem seems to be the "-mfpmath=both" compiler option, which appears not to exist.

[  314s] cd /usr/src/packages/BUILD/obj-arm-linux-gnueabihf/src/oglGraph && /usr/bin/c++   -DGLEW_STATIC -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D__WXGTK__ -I/usr/src/packages/BUILD/src/lms7002_wxgui -I/usr/src/packages/BUILD/src/LMS_Programing -I/usr/src/packages/BUILD/src/utilities_gui -I/usr/src/packages/BUILD/src/RFSpark -I/usr/src/packages/BUILD/src/FPGAcontrols_wxgui -I/usr/src/packages/BUILD/src/numericSlider -I/usr/src/packages/BUILD/src/myriad7 -I/usr/src/packages/BUILD/src/fftviewer_wxgui -I/usr/src/packages/BUILD/src/lms7suiteEvents -I/usr/src/packages/BUILD/src/boards_wxgui -isystem /usr/lib/arm-linux-gnueabihf/wx/include/gtk2-unicode-3.0 -isystem /usr/include/wx-3.0 -I/usr/src/packages/BUILD/src/oglGraph/glew -I/usr/src/packages/BUILD/src/oglGraph/glew/GL -I/usr/src/packages/BUILD/src/oglGraph  -g -O2 -fdebug-prefix-map=/usr/src/packages/BUILD=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fvisibility-inlines-hidden -pthread   -march=native -mfpmath=both -Wno-narrowing -Wall -fvisibility=hidden -std=gnu++11 -o CMakeFiles/oglGraph.dir/dlgMarkers.cpp.o -c /usr/src/packages/BUILD/src/oglGraph/dlgMarkers.cpp
[  314s] cd /usr/src/packages/BUILD/obj-arm-linux-gnueabihf/src/oglGraph && /usr/bin/c++   -DGLEW_STATIC -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D__WXGTK__ -I/usr/src/packages/BUILD/src/lms7002_wxgui -I/usr/src/packages/BUILD/src/LMS_Programing -I/usr/src/packages/BUILD/src/utilities_gui -I/usr/src/packages/BUILD/src/RFSpark -I/usr/src/packages/BUILD/src/FPGAcontrols_wxgui -I/usr/src/packages/BUILD/src/numericSlider -I/usr/src/packages/BUILD/src/myriad7 -I/usr/src/packages/BUILD/src/fftviewer_wxgui -I/usr/src/packages/BUILD/src/lms7suiteEvents -I/usr/src/packages/BUILD/src/boards_wxgui -isystem /usr/lib/arm-linux-gnueabihf/wx/include/gtk2-unicode-3.0 -isystem /usr/include/wx-3.0 -I/usr/src/packages/BUILD/src/oglGraph/glew -I/usr/src/packages/BUILD/src/oglGraph/glew/GL -I/usr/src/packages/BUILD/src/oglGraph  -g -O2 -fdebug-prefix-map=/usr/src/packages/BUILD=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fvisibility-inlines-hidden -pthread   -march=native -mfpmath=both -Wno-narrowing -Wall -fvisibility=hidden -std=gnu++11 -o CMakeFiles/oglGraph.dir/GLFont.cpp.o -c /usr/src/packages/BUILD/src/oglGraph/GLFont.cpp
[  314s] c++: error: unrecognized command line option '-mfpmath=both'
[  314s] src/oglGraph/CMakeFiles/oglGraph.dir/build.make:65: recipe for target 'src/oglGraph/CMakeFiles/oglGraph.dir/dlgMarkers.cpp.o' failed
[  314s] make[3]: *** [src/oglGraph/CMakeFiles/oglGraph.dir/dlgMarkers.cpp.o] Error 1
[  314s] make[3]: *** Waiting for unfinished jobs....
[  314s] c++: error: unrecognized command line option '-mfpmath=both'
[  314s] src/oglGraph/CMakeFiles/oglGraph.dir/build.make:89: recipe for target 'src/oglGraph/CMakeFiles/oglGraph.dir/GLFont.cpp.o' failed
[  314s] make[3]: *** [src/oglGraph/CMakeFiles/oglGraph.dir/GLFont.cpp.o] Error 1
[  314s] make[3]: Leaving directory '/usr/src/packages/BUILD/obj-arm-linux-gnueabihf'
[  314s] CMakeFiles/Makefile2:283: recipe for target 'src/oglGraph/CMakeFiles/oglGraph.dir/all' failed
[  314s] make[2]: *** [src/oglGraph/CMakeFiles/oglGraph.dir/all] Error 2
[  314s] make[2]: Leaving directory '/usr/src/packages/BUILD/obj-arm-linux-gnueabihf'
[  314s] Makefile:130: recipe for target 'all' failed
[  314s] make[1]: *** [all] Error 2
[  314s] make[1]: Leaving directory '/usr/src/packages/BUILD/obj-arm-linux-gnueabihf'
[  314s] dh_auto_build: make -j2 returned exit code 2
[  314s] debian/rules:28: recipe for target 'binary' failed
[  314s] make: *** [binary] Error 2
[  314s] dpkg-buildpackage: error: fakeroot debian/rules binary gave error exit status 2
[  314s] 
[  314s] armbuild02 failed "build limesuite_17.09.1-1.dsc" at Wed Nov 29 15:58:24 UTC 2017.
[  314s] 
[  314s] ### VM INTERACTION START ###
[  318s] [  295.888669] SysRq : Power Off
[  318s] [  295.925736] reboot: Power down
[  318s] ### VM INTERACTION END ###
[  318s] 
[  318s] armbuild02 failed "build limesuite_17.09.1-1.dsc" at Wed Nov 29 15:58:28 UTC 2017.
[  318s]

ddcc, 2017-11-29:
This seems identical to #143, which was fixed with commit f4da3e8ec73d3154072df6e35145fdcfdcd09ca6, but isn't included in 17.09.1. Have you tried 17.10.0, which does include the fix?

laf0rge, 2018-03-07:
We're using 17.12 meanwhile and it is confirmed fixed.

closed 2018-03-07 by laf0rge