myriadrf/LimeSuite #152: Missing late packet indications on the LimeSDR Mini

miek, 2017-12-02:
I'm working on a project that's doing scheduled burst TX with SoapySDR & the LimeSDR Mini and I'm having trouble debugging it because I don't seem to be getting indications of late packets.

I found that if I run my script with the LimeSDR (not-mini) I get late packet indications as expected.

This seems to have slipped past the tests in `SoapyLMS7/BasicStreamTests.py` because `readStreamStatus` always returns `SOAPY_SDR_TIME_ERROR` on the first call. If I add a dummy `readStreamStatus` to `setUp` then the test fails with the mini and continues to pass with the not-mini.

Here are the full test runs:

## Mini

```mike@geo:~/gnuradio/src/limesuite/SoapyLMS7$ python BasicStreamTests.py 
[INFO] Make connection: 'LimeSDR Mini [USB 3.0] 1D3548B7BA5942'
[INFO] Device name: LimeSDR-mini
[INFO] Reference: 40 MHz
[INFO] Init LMS7002M(0)
[INFO] Ver=7, Rev=1, Mask=1
[INFO] LMS7002M calibration values caching Disable
ret=-6, flags=4, timeNs=6000
===== receive a timed burst =====
readStream for a timed burst...
ret=412, flags=4, timeNs=100000000
ret=1024, flags=4, timeNs=100041200
ret=1024, flags=4, timeNs=100143600
ret=1024, flags=4, timeNs=100246000
ret=1024, flags=4, timeNs=100348400
ret=1024, flags=4, timeNs=100450800
ret=1024, flags=4, timeNs=100553200
ret=1024, flags=4, timeNs=100655600
ret=1024, flags=4, timeNs=100758000
ret=1024, flags=4, timeNs=100860400
ret=372, flags=6, timeNs=100962800
readStream for a timeout...
ret=-1, flags=0, timeNs=0
.[INFO] Make connection: 'LimeSDR Mini [USB 3.0] 1D3548B7BA5942'
[INFO] Device name: LimeSDR-mini
[INFO] Reference: 40 MHz
[INFO] Init LMS7002M(0)
[INFO] Ver=7, Rev=1, Mask=1
[INFO] LMS7002M calibration values caching Disable
ret=-6, flags=4, timeNs=429496729600
===== receive a burst asap =====
readStream for a burst...
ret=1024, flags=4, timeNs=0
ret=1024, flags=4, timeNs=102400
ret=1024, flags=4, timeNs=969800
ret=1024, flags=4, timeNs=1072200
ret=1024, flags=4, timeNs=1174600
ret=1024, flags=4, timeNs=1277000
ret=1024, flags=4, timeNs=1379400
ret=1024, flags=4, timeNs=1481800
ret=1024, flags=4, timeNs=1584200
ret=784, flags=6, timeNs=1686600
readStream for a timeout...
ret=-1, flags=0, timeNs=0
.[INFO] Make connection: 'LimeSDR Mini [USB 3.0] 1D3548B7BA5942'
[INFO] Device name: LimeSDR-mini
[INFO] Reference: 40 MHz
[INFO] Init LMS7002M(0)
[INFO] Ver=7, Rev=1, Mask=1
[INFO] LMS7002M calibration values caching Disable
ret=-6, flags=4, timeNs=2514404800
===== receive a continuous stream =====
readStream continuously...
readStream for a timeout...
ret=-1, flags=0, timeNs=0
.[INFO] Make connection: 'LimeSDR Mini [USB 3.0] 1D3548B7BA5942'
[INFO] Device name: LimeSDR-mini
[INFO] Reference: 40 MHz
[INFO] Init LMS7002M(0)
[INFO] Ver=7, Rev=1, Mask=1
[INFO] LMS7002M calibration values caching Disable
ret=-6, flags=4, timeNs=2513984000
===== receive a late burst =====
readStream for a late burst...
readStream for a timeout...
ret=-1, flags=0, timeNs=0
.[INFO] Make connection: 'LimeSDR Mini [USB 3.0] 1D3548B7BA5942'
[INFO] Device name: LimeSDR-mini
[INFO] Reference: 40 MHz
[INFO] Init LMS7002M(0)
[INFO] Ver=7, Rev=1, Mask=1
[INFO] LMS7002M calibration values caching Disable
[ERROR] SetFrequencyCGEN(640 MHz) failed:
INT: 63	FRAC: 0	DIV_OUTCH_CGEN: 1
VCO: 2560 MHz	RefClk: 40 MHz
TuneVCO(CGEN) - VCO too low
[ERROR] SetFrequencyCGEN(640 MHz) failed:
INT: 63	FRAC: 0	DIV_OUTCH_CGEN: 1
VCO: 2560 MHz	RefClk: 40 MHz
TuneVCO(CGEN) - VCO too low
[ERROR] SetFrequencyCGEN(640 MHz) failed:
INT: 63	FRAC: 0	DIV_OUTCH_CGEN: 1
VCO: 2560 MHz	RefClk: 40 MHz
TuneVCO(CGEN) - VCO too low
ret=-6, flags=4, timeNs=439381858730900
===== test the timestamps =====
t0=0 ns
t1=101133000 ns
delta=1.011330 secs
.[INFO] Make connection: 'LimeSDR Mini [USB 3.0] 1D3548B7BA5942'
[INFO] Device name: LimeSDR-mini
[INFO] Reference: 40 MHz
[INFO] Init LMS7002M(0)
[INFO] Ver=7, Rev=1, Mask=1
[INFO] LMS7002M calibration values caching Disable
ret=-6, flags=4, timeNs=2513529600
===== test txing a late packet =====
readStreamStatus for a late indicator...
F
======================================================================
FAIL: testTxLate (__main__.TestBasicStreaming)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "BasicStreamTests.py", line 171, in testTxLate
    self.assertEqual(r0.ret, SOAPY_SDR_TIME_ERROR)
AssertionError: -1 != -6

----------------------------------------------------------------------
Ran 6 tests in 36.933s

FAILED (failures=1)
```

## not-mini

```mike@geo:~/gnuradio/src/limesuite/SoapyLMS7$ python BasicStreamTests.py 
[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 9060B00462320'
[INFO] Reference clock 30.720 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] Init LMS7002M(0)
[INFO] Ver=7, Rev=1, Mask=1
[INFO] LMS7002M calibration values caching Disable
ret=-6, flags=4, timeNs=170166923449
===== receive a timed burst =====
readStream for a timed burst...
[INFO] L
ret=946, flags=4, timeNs=100000000
ret=1024, flags=4, timeNs=100094600
ret=1024, flags=4, timeNs=100197000
ret=1024, flags=4, timeNs=100299400
ret=1024, flags=4, timeNs=100401800
ret=1024, flags=4, timeNs=100504200
ret=1024, flags=4, timeNs=100606600
ret=1024, flags=4, timeNs=100709000
ret=1024, flags=4, timeNs=100811400
ret=862, flags=6, timeNs=100913800
readStream for a timeout...
ret=-1, flags=0, timeNs=0
.[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 9060B00462320'
[INFO] Reference clock 30.720 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] Init LMS7002M(0)
[INFO] Ver=7, Rev=1, Mask=1
[INFO] LMS7002M calibration values caching Disable
ret=-6, flags=4, timeNs=2490126409
===== receive a burst asap =====
readStream for a burst...
ret=1024, flags=4, timeNs=0
ret=1024, flags=4, timeNs=102400
ret=1024, flags=4, timeNs=204800
ret=1024, flags=4, timeNs=307200
ret=1024, flags=4, timeNs=409600
ret=1024, flags=4, timeNs=512000
ret=1024, flags=4, timeNs=1175400
ret=1024, flags=4, timeNs=1277800
ret=1024, flags=4, timeNs=1380200
ret=784, flags=6, timeNs=1482600
readStream for a timeout...
ret=-1, flags=0, timeNs=0
.[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 9060B00462320'
[INFO] Reference clock 30.720 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] Init LMS7002M(0)
[INFO] Ver=7, Rev=1, Mask=1
[INFO] LMS7002M calibration values caching Disable
ret=-6, flags=4, timeNs=2348235209
===== receive a continuous stream =====
readStream continuously...
readStream for a timeout...
ret=-1, flags=0, timeNs=0
.[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 9060B00462320'
[INFO] Reference clock 30.720 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] Init LMS7002M(0)
[INFO] Ver=7, Rev=1, Mask=1
[INFO] LMS7002M calibration values caching Disable
ret=-6, flags=4, timeNs=13770791387831
===== receive a late burst =====
readStream for a late burst...
[INFO] L
readStream for a timeout...
ret=-1, flags=0, timeNs=0
.[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 9060B00462320'
[INFO] Reference clock 30.720 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] Init LMS7002M(0)
[INFO] Ver=7, Rev=1, Mask=1
[INFO] LMS7002M calibration values caching Disable
ret=-6, flags=4, timeNs=2479348809
===== test the timestamps =====
t0=0 ns
t1=100113000 ns
delta=1.001130 secs
[INFO] L
.[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 9060B00462320'
[INFO] Reference clock 30.720 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] Init LMS7002M(0)
[INFO] Ver=7, Rev=1, Mask=1
[INFO] LMS7002M calibration values caching Disable
ret=-6, flags=4, timeNs=2478459209
===== test txing a late packet =====
[INFO] L
readStreamStatus for a late indicator...
delta=0.105111 secs
readStreamStatus for a timeout...
[INFO] L
.
----------------------------------------------------------------------
Ran 6 tests in 41.088s

OK
```

guruofquality, 2017-12-04:
> This seems to have slipped past the tests in SoapyLMS7/BasicStreamTests.py because readStreamStatus always returns SOAPY_SDR_TIME_ERROR on the first call.

@ztamosevicius This is the spurious late packet I mentioned, I tracked it as far as: https://github.com/myriadrf/LimeSuite/blob/master/src/ConnectionSTREAM/ConnectionSTREAMing.cpp#L287 Any idea?

ztamosevicius, 2017-12-14:
Update the gateware to the latest one (at the time or writing):
https://github.com/myriadrf/LimeSDR-Mini_GW/tree/master/LimeSDR-Mini_bitstreams
The gateware was not reporting late packets.

closed 2018-03-29 by IgnasJarusevicius