myriadrf/LimeSuite #153: Current version of Limesuite does not seem to allow initiating stream on 2nd channel exclusively.

brendanthebig, 2017-12-10:
Please see below link for more info.

https://github.com/f4exb/sdrangel/issues/101

IgnasJarusevicius, 2017-12-11:
should be fixed with 04b57e0a247debb451b067b3fc486307627ba5b7

brendanthebig, 2017-12-11:
Thanks @IgnasJarusevicius

vatrat, 2018-01-28:
Should this be closed? @brendanthebig

KarlL2, 2018-03-22:
I guess it should be closed. I also noticed that issue previously where I couldn't receive on channel 1 if channel 0 wasn't receiving also (or maybe if it wasn't setup). Now I can correctly receive only on channel 1 using SoapySDR.

In https://github.com/f4exb/sdrangel/issues/101 they also agreed it's working fine now.

brendanthebig, 2018-03-22:
Sorry, haven't been back on this site for a while. Yes, everything is working well now...Thanks to all involved.

Regards,
Brendan Jenkins,
VK3WWB.

closed 2018-03-22 by brendanthebig