myriadrf/LimeSuite #156: 1 PPS timestamps synchronization

ccsh23, 2017-12-22:
Dear all,

From the Ignas post [here](https://github.com/myriadrf/LimeSuite/issues/149#issuecomment-352750275) it looks like one very heplfun enhancement for LimeSDR has recently been implemented (TX/RX state signallization via GPIO pin), I just wanted to know how are your thoughts about other very often raised enhancement: timestamp synchronization via external 1 PPS signal. As for today LimeSDR competes mostly with USRP B210 and XTRX devices, both of which have that feature implemented. So I would like to know what do you think about that: is it even possible to use another GPIO pin for that purpose, are you considering implementing this in future, or are already sure that it is not planned etc.

I would be greateful for any information.

Best regards!

laf0rge, 2018-04-28:
I'd also be intrested to hear about this capability.

9600, 2018-05-02:
@ztamosevicius, do you know if this is on the roadmap?

ccsh23, 2018-05-18:
Aby update?

avkost, 2019-01-21:
Hi. I am also very interested in this kind of enhancement.