myriadrf/LimeSuite #157: segfault when launching LimeSuiteGUI

c5e3, 2017-12-25:
Version: limesuite17.12.0-myriadrf~artful, liblimesuite17.12.0-myriadrf~artful)

the problem occurs, when launching LimeSuiteGUI or any application that uses gr-osmosdr + soapy

there are not other error messages than this:
´traps: LimeSuiteGUI[23171] trap invalid opcode ip:55b59092a78f sp:7fff70dd8fd0 error:0 in LimeSuiteGUI[55b590803000+2f3000]´

CPU: Intel i5-7400. limesuite works perfectly on a i5-5300U though
OS: Ubuntu 17.10
kernel: 4.13.0-21-generic

arodland, 2017-12-31:
I can confirm, I upgraded to 17.12.0-myriadrf~artful myself yesterday and everything that tries to use the radio crashes with an illegal instruction error. An easy reproducer is "LimeUtil --timing".

```
andrew@goron:~$ gdb LimeUtil     
GNU gdb (Ubuntu 8.0.1-0ubuntu1) 8.0.1
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from LimeUtil...(no debugging symbols found)...done.
(gdb) set args --timing
(gdb) run
Starting program: /usr/bin/LimeUtil --timing
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff62e2700 (LWP 27446)]
[New Thread 0x7ffff5ae1700 (LWP 27447)]
[New Thread 0x7ffff52e0700 (LWP 27448)]
Connected to [LimeSDR-USB, media=USB 3.0, module=STREAM, addr=1d50:6108, serial=0009060B00471F18]
Estimated reference clock 30.7198 MHz
Reference clock 30.720 MHz
Creating instance of LMS7002M:

Timing basic operations:
  >>> SPI write register:       137.237 us
  >>> SPI read register:        138.752 us

Thread 1 "LimeUtil" received signal SIGILL, Illegal instruction.
0x00007ffff7b500d5 in lime::LMS7002M::GetFrequencyCGEN() () from /usr/lib/x86_64-linux-gnu/libLimeSuite.so.17.12-1
(gdb) bt
#0  0x00007ffff7b500d5 in lime::LMS7002M::GetFrequencyCGEN() () from /usr/lib/x86_64-linux-gnu/libLimeSuite.so.17.12-1
#1  0x00007ffff7b50100 in lime::LMS7002M::GetReferenceClk_TSP(bool) () from /usr/lib/x86_64-linux-gnu/libLimeSuite.so.17.12-1
#2  0x00007ffff7b5033e in lime::LMS7002M::SetNCOFrequency(bool, unsigned char, double) () from /usr/lib/x86_64-linux-gnu/libLimeSuite.so.17.12-1
#3  0x0000555555559bb7 in ?? ()
#4  0x0000555555558815 in ?? ()
#5  0x00007ffff6f9b1c1 in __libc_start_main (main=0x555555557170, argc=2, argv=0x7fffffffda98, init=<optimized out>, fini=<optimized out>, 
    rtld_fini=<optimized out>, stack_end=0x7fffffffda88) at ../csu/libc-start.c:308
#6  0x0000555555558d7a in _start ()
(gdb) 
```

Given the location of the crash, I have some suspicion of https://github.com/myriadrf/LimeSuite/commit/12108cf5af6274e938e9a78597a459258dd4496a being involved, but also maybe compiler flags? It's an odd one.

guruofquality, 2017-12-31:
I'm seeing this too, definitely odd. But its not happening when I build the same debs manually on artful. It may be a flags issue or something odd in the PPA build environment. I will probably try rebuilding just in case it was an accident on a particular PPA build machine.

guruofquality, 2018-01-01:
So `-march=native` can be issue when packaging for x86 when you want the binaries to work across many SIMD architectures. This commit should address this issue: https://github.com/myriadrf/LimeSuite/commit/305ac5d3aed5f0f280d73860dcbbc40d63b3d6ee and I have uploaded a new build for artful. Happy new year!

arodland, 2018-01-01:
Thank you and happy 2018! I upgraded using apt to 17.12.0.4.305ac5d-myriadrf1~artful and I can confirm everything working fine.

vatrat, 2018-01-28:
Should this be closed because of commit 305ac5d? @arodland @guruofquality

guruofquality, 2018-02-01:
Yea, it was a tricky bug, but we should be good going forward.

closed 2018-02-01 by guruofquality