myriadrf/LimeSuite #166: Linux, macOS packages do not create menu entries

ghalfacree, 2018-02-21:
The Linux (Ubuntu PPA) and macOS (Homebrew) packages for LimeSuite do not create menu entries for the Lime Suite GUI, making the only way to find and run the software post-installation via the command line.

For user friendliness, the packages should be modified to create menu entries for the Lime Suite GUI. Ideally, the same should also be done to the makefile for when Lime Suite is compiled from source.

ghalfacree, 2018-02-21:
Pinging @IgnasJarusevicius on this, per @9600's recommendation.

guruofquality, 2018-02-23:
Odd, for the debian packages its got a postinst: https://github.com/myriadrf/LimeSuite/blob/master/debian/limesuite.postinst and doing `sudo apt-get install limesuite` gives me:

```
...
Moving images to /usr/share/LimeSuite/images/17.12
Setting up limesuite-udev:amd64 (17.12.0.4.305ac5d-myriadrf1~artful) ...
Setting up limesuite-images:amd64 (17.12.0.4.305ac5d-myriadrf1~artful) ...
Setting up limesuite (17.12.0.4.305ac5d-myriadrf1~artful) ...
Install Lime Suite icon resources...    OK
Install Lime Suite menu launcher...     OK
```

and the menu shows up in kde (which I tested on)

As for osx, there are some build changes for MACOSX_BUNDLE stuff in cmake. I can post this here for reference because it took me a while to get it right the first time I need to do something like this: https://github.com/pothosware/PothosFlow/blob/master/CMakeLists.txt#L209

FFY00, 2018-06-13:
I agree with this. I had to manually create a `.desktop` file when I added limesuite to the Arch Linux AUR repository. Optimally, this should be done with Cmake.

In the meantime people can install my `.desktop` file manually if they want.
https://aur.archlinux.org/cgit/aur.git/tree/limesuitegui.desktop?h=limesuite