myriadrf/LimeSuite #167: Silence non-critical compiler warnings

ghalfacree, 2018-02-21:
Compiling Lime Suite from source on any supported platform results in a large number of non-critical warnings from the compiler. While these do not affect the compilation or functionality of the software in any way, it would be nice if they could be suppressed so as not to worry newcomers to the software.

f4exb, 2018-03-17:
Seconding ghalfacree on this issue. Warnings are usually there for a good reason so I would welcome some code cleanup so that these warnings do not appear anymore. Some of these warnings might appear benign but they may reveal some more serious issues.

orinem, 2018-05-24:
Guess I shouldn't have raised a similar issue for OSX then...

But there was a probably error hiding in all the noise which I reported separately.

I find the double to float loss of precision warnings for temporary stack variables particularly annoying since the next time the float in question gets used, it gets promoted back to double anyway.

FFY00, 2018-06-13:
Are you sure this is not due to the use of old versions of the dependencies? In my up-to-date Arch system, the only warnings I get are related to the use of deprecated libusb functions, which should be fixed by #196.

IgnasJarusevicius, 2018-12-11:
Number of compiler warning depends on compiler. I am not seeing a lot of warnings on Ubuntu 16.04 with gcc 5.4.0, however I think MSVC produces quite a lot of them (I rarely use it). There is no plans to do any work specifically to remove warnings, however if a warning pop-up in a file that I am editing for other reasons I usually try to remove it.

closed 2018-12-11 by IgnasJarusevicius