myriadrf/LimeSuite #170: Lack of pkg-config support

laf0rge, 2018-03-07:
One thing I noticed immediately when starting to use LimeSuite is the lack of pkg-config support.

pkg-config is for decades the standard method on Linux by which an application can determine the given platform-specific compiler and linker flags required to compile + link against any library.  It's very
easy to integrate into whatever build system (CMake, autoconf, ...)

Not having pkg-config support means that every application has to jump through various hoops and have its own rules regarding which paths and flags to use for lime suite on each and any of the supported platforms (Linux distributions/dialects/architectures).  I'm wondering why LimeSuite is putting this burden on their application developers.  It just means that everyone will end up having a different set of autoconf macros and defaults, and that portability of those applications suffers.

9600, 2018-03-07:
@IgnasJarusevicius could we add support for pkg-config, please.

laf0rge, 2018-04-28:
confirmed fixed.

closed 2018-04-28 by laf0rge