myriadrf/LimeSuite #171: lms7 - bad freqency reported for TX and RX

neeocf, 2018-03-12:
as of commit c02f4358 API returns bad values for RX and TX frequency ranges supported by device (and it broke for example sdrangel). please apply following patch to fix this issue:
```
diff --git a/src/API/lms7_device.cpp b/src/API/lms7_device.cpp
index c957dd7..b576ce1 100644
--- a/src/API/lms7_device.cpp
+++ b/src/API/lms7_device.cpp
@@ -748,7 +748,7 @@ LMS7_Device::Range LMS7_Device::GetTxPathBand(unsigned path, unsigned chan) cons
 {
   switch (path)
   {
-      case LMS_PATH_TX2: return Range(2e6, 3.8e6);
+      case LMS_PATH_TX2: return Range(2e9, 3.8e9);
       case LMS_PATH_TX1: return Range(30e3, 2e9);
       default: return Range();
   }
@@ -1291,7 +1291,7 @@ double LMS7_Device::GetFrequency(bool tx, unsigned chan) const
 
 LMS7_Device::Range LMS7_Device::GetFrequencyRange(bool tx) const
 {
-  return Range(30e6, 3.8e6);
+  return Range(30e6, 3.8e9);
 }
 
 int LMS7_Device::Init()
```

IgnasJarusevicius, 2018-03-13:
patch applied 67d88c54690eecf6b73f0a06ab5e58429926bc6d

neeocf, 2018-03-13:
ok, but souldn't it actually be 100e3 - 3.8e9 for limesdr and 10e6 - 3.5e9 for the mini?

I've also noticed that LMS_PATH_TX1 above is off as well - it's 30e3 and not 30e6.

rjonaitis, 2018-04-17:
Already fixed in 29f8f4c39b8a0

closed 2018-04-17 by rjonaitis