myriadrf/LimeSuite #173: Mess with LMS7_Device::SetFrequency and NCO

f4exb, 2018-03-16:
Recent changes (I think from commit c0f345) have introduced a dependency between LO frequency and NCO for channels A and B. This effectively prevents the control of the NCO by third party applications.

I don't know what is the purpose behind this choice but there is no reason to tie these controls with one another. If I read the specs of LMS7002M the LO is common between channels A and B and there is a NCO for each channel and these NCOs can be controlled independently. This is causing a mess in client applications (namely SDRangel) that expect LimeSuite to follow the hardware specs which should be the case for a low level API.

Please make these controls independent again. Else provide a way to effectively control LO and NCO independently.

f4exb, 2018-03-16:
Correction: it seems to be commit afc71d0 "Allow set different frequencies for A/B channels (experimental)". As the title says this should remain "experimental"...

f4exb, 2018-03-16:
It appears that center frequency and sample rate is now independent between channels A and B. How is that possible?

f4exb, 2018-03-17:
These are the API calls I make to set LO frequency or sample rate:
  - LMS_SetLOFrequency 
  - LMS_SetSampleRateDir

Also made the title of the issue more clear.

IgnasJarusevicius, 2018-03-19:
Now (in master) LMS_SetLOFrequency() attempts to set different frequencies for A and B channels using NCO when possible, previously (in stable branch) it was using NCO only for frequencies bellow 30 MHz. This is to allow setting different frequencies in applications that only use high level functions (frequency, sample rate, gain ,bandwidth). To set SXT/SXR frequency only, you can use low level function LMS_SetClockFreq() with LMS_CLOCK_SXR / LMS_CLOCK_SXT clock parameters.

I dont't think that functionality of LMS_SetSampleRateDir() has recently changed. It just attempts to set different sample rates for Tx and Rx by varying CGEN divider and interpolation/decimation ratios.

f4exb, 2018-03-20:
Using LMS_SetClockFreq() with LMS_CLOCK_SXR / LMS_CLOCK_SXT clock parameters in place of LMS_SetLOFrequency() does the trick. 

Please maintain this API in the future as it is essential to control the LMS7002M finely. I expect my users to know how the LimeSDR hardware works and want them to have full control.

I haven't  noticed the sample rate issue anymore maybe I was mistaken. You can close this issue I will open a new one if I have more or any evidence.

Thank you and best regards.

closed 2018-03-29 by IgnasJarusevicius