myriadrf/LimeSuite #177: LimeSDR USB and LimeSDR Mini freeze with latest update

juribeparada, 2018-03-26:
Hi, I compiled the latest LimeSuite (with latest GW), and when I try to use for receiving, both devices stop sending data when I used with CubicSDR and high sample rate (> 2 MSPS for example). However, if I downgrade to GW 2.14 in LimeSDR USB this doesn't happen (even with LimeSuite 18.03). I realized with LimeSDR Mini, TX works very well now with latest LimeSuite and GW, but RX freeze (CubicSDR and high sample rate, as LimeSDR USB). This problem with LimeSDR Mini happens with commit 23ae40e, with the modification of LMS7_LimeSDR_mini::SetRate() function. I tested under Linux Ubuntu 16.04 and macOS.

hewittc, 2018-06-06:
Same symptoms here on Arch Linux where > 2 MS/s RX results in read errors.

> [INFO] Make connection: 'LimeSDR Mini [USB 3.0] 1D398DACB0EBDB'
> [INFO] Reference clock 40.00 MHz
> [INFO] Device name: LimeSDR-Mini
> [INFO] Reference: 4e+07 MHz
> [INFO] LMS7002M calibration values caching Disable
> SDR thread starting.
> device init()
> Device Stream MTU: 4080
> [INFO] Rx calibration finished
> starting readLoop()
> SDRThread::readStream(): 2. SoapySDR read failed with code: -1
> SDRThread::readStream(): 2. SoapySDR read failed with code: -1
> SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...

juribeparada, 2019-03-09:
UPDATE:
Today I compiled latest versions from GitHub and tested under macOS Mojave (10.14.3):
- LimeSuite (19.01)
- SoapySDR (0.8.0)
- CubicSDR (0.2.5)
And using latest GW:
- LimeSDR-USB (1.29)
- LimeSDR-Mini (2.20)

LimeSDR-Mini now working fine until 5 MSPS (actually stops working with >= 5 MSPS), not bad for Mini version anyway. I noted this in:
https://github.com/myriadrf/LimeSuite/blob/e3e7203ccf1e52128950347d44eb4e11a0ac7088/src/FPGA_common/FPGA_Mini.cpp#L71
If I increase these numbers, I can work higher sample rates for LimeSDR-Mini with CubicSDR. Why this limitation?

For LimeSDR-USB, this patch solves the problem in CubicSDR: https://github.com/cjcliffe/CubicSDR/commit/ed5993748fc48bd2da340f8bc2ef240513e777f3

rjonaitis, 2019-03-09:
As mentioned in #251 if CubicSDR is really doing sample rate change while the samples streaming is active, it will most likely break streaming.
Sample rate change results in couple of hardware clocks and interface frequency changes, doing that while they are being used for data transfering can put FPGA/USB pipeline in undefined state. So stream would need to be stopped and restarted to get it back in proper state.

```If I increase these numbers, I can work higher sample rates for LimeSDR-Mini with CubicSDR. Why this limitation?```
It's not a limitation on sampling rate, it's just as the sample rate increases the clocks phases needs to be configured more precisely to ensure correct behaviour. There is no clear hard limit when such adjustements needs to be done, therefore it still works when you increase the limit for phase search triggering. Triggering clock phase adjustment while data is streaming will definitely break it as mentioned above.

juribeparada, 2019-03-10:
Thank you for the explanation. Then my original issue was always related to the way that CubicSDR handles the sample rate change.
I can close this issue now.

closed 2019-03-10 by juribeparada