myriadrf/LimeSuite #184: osmo-trx unable to use limesdr through uhd in current master after LimeSuite v18.04.1

pespin, 2018-04-25:
LimeSuite v18.04.1 (a2d7d6ac547eb0500a7de8d5e0030a6892789951) works fine, device can be started by osmo-trx using UHD without problem.

However, when trying with current master Limesuite (b1f38113328fa5128adf11fc6e54898cbfea4a78, 6 commits after latest working release), osmo-trx is unable to open the device through UHD.

It fails with following messages:

```
osmo-trx/CommonLibs/Logger.cpp:53 [tid=140550775385792] Setting tx bw 5e+06 for 0

UHD Error:
    Tune Tx Filter: MCU error 14 (Tx RCAL_LPF range limit reached,)

UHD Error:
    setBandwidth(Tx, 0, 5 MHz) Failed -
terminate called after throwing an instance of 'std::runtime_error'
  what():
signal 6 received
```

rjonaitis, 2018-05-02:
Set Tx filter bandwidth to a bit more than 5 MHz, like 5.2 MHz.

pespin, 2018-05-08:
Thanks, using LMS_SetLPFBW(m_lms_dev, LMS_CH_TX, i, 5.2e6) instead of 5e6 fixed the issue. 

However, I find it a bit strange that it is not catched by the more understandable error report when the check against TxLPF_RF_LimitLow (5e6) is done in LMS7002M::TuneTxFilter. I think it should be catched there somehow if 5MHz is known to fail. What do you think?

pespin, 2018-05-25:
Today I tested again my LimeSDR and it's again failing after applying the changes to use 5.2 MHz. Changing it to 5.3MHz made it work again.

commit cd165941 (2018-06-04): Relax Tx filter calibration lower bound check, fix #184

Due to board differences when calibrating Tx filter close to 5 MHz bandwidth, sometimes the measured signal might be very close but not pass the expected threshold, so if filter bandwidth lower bound is reached consider it successfull
closed 2018-06-04 by rjonaitis