myriadrf/LimeSuite #186: Strange timestamp behavior with osmo-trx-lms and LMS_RecvStream

pespin, 2018-04-26:
I am currently in the process of writing a new direct backend for osmo-trx which uses LimeSuite directly instead of going through the entire stack of UHD->SoapyUHD->SoapySDR->SoapyLMS7->LimeSuite.

Branch adding initial support for it can be found in upstream branch pespin/lime:
https://git.osmocom.org/osmo-trx/log/?h=pespin/lime

However, I am currently seeing some unexpected behavior with timestamps.

Basically, I call LMS_RecvStream in a tight loop with len=2500, and first times the jump in timestamp after each iteration is 2500 (0x9c4) as expected. However, at the 4th iteration the jump is of  4d1c - 1388 = 14740 samples, so it seems something in there is not going as I would expect.

```
20180426144040908 DMAIN <0000> LMSDevice.cpp:326 [tid=140607662315264] Flush: Recv buffer of len 2500 at 0
20180426144040908 DMAIN <0000> LMSDevice.cpp:326 [tid=140607662315264] Flush: Recv buffer of len 2500 at 9c4
20180426144040909 DMAIN <0000> LMSDevice.cpp:326 [tid=140607662315264] Flush: Recv buffer of len 2500 at 1388
20180426144040909 DMAIN <0000> LMSDevice.cpp:326 [tid=140607662315264] Flush: Recv buffer of len 2500 at 4d1c
20180426144040909 DMAIN <0000> LMSDevice.cpp:326 [tid=140607662315264] Flush: Recv buffer of len 2500 at 56e0
20180426144040909 DMAIN <0000> LMSDevice.cpp:326 [tid=140607662315264] Flush: Recv buffer of len 2500 at 60a4
20180426144040910 DMAIN <0000> LMSDevice.cpp:326 [tid=140607662315264] Flush: Recv buffer of len 2500 at 6a68
20180426144040910 DMAIN <0000> LMSDevice.cpp:326 [tid=140607662315264] Flush: Recv buffer of len 2500 at 742c
20180426144040910 DMAIN <0000> LMSDevice.cpp:326 [tid=140607662315264] Flush: Recv buffer of len 2500 at 7df0
20180426144040910 DMAIN <0000> LMSDevice.cpp:326 [tid=140607662315264] Flush: Recv buffer of len 2500 at 87b4
20180426144040911 DMAIN <0000> LMSDevice.cpp:326 [tid=140607662315264] Flush: Recv buffer of len 2500 at 9178
20180426144040911 DMAIN <0000> LMSDevice.cpp:326 [tid=140607662315264] Flush: Recv buffer of len 2500 at 9b3c
```

Further information for this report can be found in:
https://osmocom.org/issues/2919#note-2

9600, 2018-05-02:
@IgnasJarusevicius, would be grateful if you could take a look at this, please.

rjonaitis, 2018-05-02:
https://git.osmocom.org/osmo-trx/tree/Transceiver52M/device/lms/LMSDevice.cpp?h=pespin/lime#n143
The main problem I see with your code is that you first do your configuration, and then you call `LMS_Init`, which basicaly resets the board and all your configuration is lost. That would also explain your note `output from ettus b200 running with UHD suggests that the limesdr clock is running around 4 times faster than it should.`

After you `LMS_Open` then you should call `LMS_Init` and after that you can apply your desired settings.

pespin, 2018-05-02:
Indeed, we already spotted and fixed that issue in https://git.osmocom.org/osmo-trx/log/Transceiver52M/device/lms/LMSDevice.cpp?h=laforge/lime, however I'm not sure how that can explain the jumping in the timestamps.
I'll do some more tests soon with the new code in this branch and provide feedback regarding this issue.

rjonaitis, 2018-05-02:
check out these threads:
https://discourse.myriadrf.org/t/small-packet-loss-in-the-beginning-of-the-rx-stream/1861/2

https://discourse.myriadrf.org/t/limesdr-tx-and-unknown-features-in-rx/2652/2

IgnasJarusevicius, 2018-05-04:
I have done some testing with LimeSDR-USB on Ubuntu 16.04 by adding some timestamps checks and printf's to LimeSutieGUI and dualRXTX example. For me timestamps start at 0 and continues without jumps so USB buffers seem to be properly cleared on stream start.
Normally, timestamps  jumps occur when samples are not read from HW fast enough.

pespin, 2018-05-04:
Yes, maybe that happens because we were calling Init() at the end and maybe a really high sample rate was set. I'll give it a try with our new branch soon and provide feedback on this issue.

pespin, 2018-05-08:
Indeed, with our last changes calling Init at proper place I no longer see the behavior I posted here. Closing the issue.

closed 2018-05-08 by pespin