myriadrf/LimeSuite #188: bulid armhf64 error 4/28/2018

KC7NOA, 2018-04-29:
[ 80%] Building CXX object src/CMakeFiles/LimeSuiteGUI.dir/lms7suiteEvents/lms7suiteEvents.cpp.o
[ 81%] Building CXX object src/CMakeFiles/LimeSuiteGUI.dir/boards_wxgui/pnlQSpark.cpp.o
[ 81%] Building CXX object src/CMakeFiles/LimeSuiteGUI.dir/boards_wxgui/pnluLimeSDR.cpp.o
[ 82%] Building CXX object src/CMakeFiles/LimeSuiteGUI.dir/boards_wxgui/pnlUltimateEVB.cpp.o
[ 83%] Building CXX object src/CMakeFiles/LimeSuiteGUI.dir/boards_wxgui/pnlBuffers.cpp.o
[ 84%] Building C object src/CMakeFiles/LimeSuiteGUI.dir/__/external/kissFFT/kiss_fft.c.o
[ 84%] Building CXX object src/CMakeFiles/LimeSuiteGUI.dir/windowFunction.cpp.o
[ 85%] Building CXX object src/CMakeFiles/LimeSuiteGUI.dir/boards_wxgui/pnlLimeSDR.cpp.o
[ 86%] Linking CXX executable ../bin/LimeSuiteGUI
libLimeSuite.so.18.04.1: undefined reference to `__loadConnectionNovenaRF7Entry()'
collect2: error: ld returned 1 exit status
src/CMakeFiles/LimeSuiteGUI.dir/build.make:1553: recipe for target 'bin/LimeSuiteGUI' failed
make[2]: *** [bin/LimeSuiteGUI] Error 1
CMakeFiles/Makefile2:136: recipe for target 'src/CMakeFiles/LimeSuiteGUI.dir/all' failed
make[1]: *** [src/CMakeFiles/LimeSuiteGUI.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
kc7noaa@odroid64:~/Downloads/LimeSuite/build$ sudo make clean
[sudo] password for kc7noaa: 
kc7noaa@odroid64:~/Downloads/LimeSuite/build$ cd ../..
kc7noaa@odroid64:~/Downloads$ cd LimeSuite
kc7noaa@odroid64:~/Downloads/LimeSuite$ git pull
Already up-to-date.
kc7noaa@odroid64:~/Downloads/LimeSuite$ git branch

IgnasJarusevicius, 2018-04-29:
Have you tried removing/cleaning build directory and re-running CMake?

KC7NOA, 2018-04-29:
Yea .... as a normal procedure

rjonaitis, 2018-04-29:
@IgnasJarusevicius https://github.com/myriadrf/LimeSuite/commit/c02f4358b6eb3a83f333e8951419f34b18667dde#diff-95e351a3805a1dafa85bf20b81d086e6R148

@KC7NOA delete your `build` directory and recreate it, it might be an ENABLE_NOVENARF7 define flag left enabled in the cmake cache from older version.

KC7NOA, 2018-05-03:
yea ... must something like that ... new folder worked

closed 2018-05-03 by rjonaitis