myriadrf/LimeSuite #189: build error raspberry

aligrt, 2018-04-29:
I have following on Linking libLMS7Support.

Any one can help/guide? 

[100%] Linking CXX shared module libLMS7Support.so
CMakeFiles/LMS7Support.dir/Settings.cpp.o: In function `SoapyLMS7::getGainRange(int, unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const':
Settings.cpp:(.text+0x8f8): undefined reference to `SoapySDR::Device::getGainRange(int, unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) cons                                           t'
CMakeFiles/LMS7Support.dir/Settings.cpp.o:(.data.rel.ro+0x1c): undefined reference to `SoapySDR::Device::setFrontendMapping(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::all                                           ocator<char> > const&)'
CMakeFiles/LMS7Support.dir/Settings.cpp.o:(.data.rel.ro+0x20): undefined reference to `SoapySDR::Device::getFrontendMapping[abi:cxx11](int) const'
CMakeFiles/LMS7Support.dir/Settings.cpp.o:(.data.rel.ro+0x28): undefined reference to `SoapySDR::Device::getChannelInfo[abi:cxx11](int, unsigned int) const'
CMakeFiles/LMS7Support.dir/Settings.cpp.o:(.data.rel.ro+0xd8): undefined reference to `SoapySDR::Device::setFrequency(int, unsigned int, double, std::map<std::__cxx11::basic_string<char, std::ch                                           ar_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::                                           allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std:                                           :allocator<char> > > > > const&)'
CMakeFiles/LMS7Support.dir/Settings.cpp.o:(.data.rel.ro+0x124): undefined reference to `SoapySDR::Device::listClockSources[abi:cxx11]() const'
CMakeFiles/LMS7Support.dir/Settings.cpp.o:(.data.rel.ro+0x128): undefined reference to `SoapySDR::Device::setClockSource(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<c                                           har> > const&)'
CMakeFiles/LMS7Support.dir/Settings.cpp.o:(.data.rel.ro+0x12c): undefined reference to `SoapySDR::Device::getClockSource[abi:cxx11]() const'
CMakeFiles/LMS7Support.dir/Settings.cpp.o:(.data.rel.ro+0x130): undefined reference to `SoapySDR::Device::listTimeSources[abi:cxx11]() const'
CMakeFiles/LMS7Support.dir/Settings.cpp.o:(.data.rel.ro+0x134): undefined reference to `SoapySDR::Device::setTimeSource(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<ch                                           ar> > const&)'
CMakeFiles/LMS7Support.dir/Settings.cpp.o:(.data.rel.ro+0x138): undefined reference to `SoapySDR::Device::getTimeSource[abi:cxx11]() const'
CMakeFiles/LMS7Support.dir/Settings.cpp.o:(.data.rel.ro+0x148): undefined reference to `SoapySDR::Device::setCommandTime(long long, std::__cxx11::basic_string<char, std::char_traits<char>, std::                                           allocator<char> > const&)'
CMakeFiles/LMS7Support.dir/Settings.cpp.o:(.data.rel.ro+0x178): undefined reference to `SoapySDR::Device::writeRegisters(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<c                                           har> > const&, unsigned int, std::vector<unsigned int, std::allocator<unsigned int> > const&)'
CMakeFiles/LMS7Support.dir/Settings.cpp.o:(.data.rel.ro+0x17c): undefined reference to `SoapySDR::Device::readRegisters(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<ch                                           ar> > const&, unsigned int, unsigned int) const'
CMakeFiles/LMS7Support.dir/Settings.cpp.o:(.data.rel.ro+0x188): undefined reference to `SoapySDR::Device::readSetting(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char                                           > > const&) const'
CMakeFiles/LMS7Support.dir/Settings.cpp.o:(.data.rel.ro+0x194): undefined reference to `SoapySDR::Device::readSetting(int, unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>,                                            std::allocator<char> > const&) const'
CMakeFiles/LMS7Support.dir/Settings.cpp.o:(.data.rel.ro+0x1a0): undefined reference to `SoapySDR::Device::writeGPIO(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>                                            > const&, unsigned int, unsigned int)'
CMakeFiles/LMS7Support.dir/Settings.cpp.o:(.data.rel.ro+0x1ac): undefined reference to `SoapySDR::Device::writeGPIODir(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<cha                                           r> > const&, unsigned int, unsigned int)'
CMakeFiles/LMS7Support.dir/Settings.cpp.o:(.data.rel.ro+0x1c0): undefined reference to `SoapySDR::Device::listUARTs[abi:cxx11]() const'
CMakeFiles/LMS7Support.dir/Settings.cpp.o:(.data.rel.ro+0x1c4): undefined reference to `SoapySDR::Device::writeUART(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>                                            > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
CMakeFiles/LMS7Support.dir/Settings.cpp.o:(.data.rel.ro+0x1c8): undefined reference to `SoapySDR::Device::readUART(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >                                            const&, long) const'
CMakeFiles/LMS7Support.dir/Registration.cpp.o: In function `limeSuiteLogHandler(lime::LogLevel, char const*)':
Registration.cpp:(.text+0x134): undefined reference to `SoapySDR::log(SoapySDRLogLevel, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
Registration.cpp:(.text+0x1a4): undefined reference to `SoapySDR::log(SoapySDRLogLevel, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
Registration.cpp:(.text+0x20c): undefined reference to `SoapySDR::log(SoapySDRLogLevel, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
Registration.cpp:(.text+0x274): undefined reference to `SoapySDR::log(SoapySDRLogLevel, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
Registration.cpp:(.text+0x2dc): undefined reference to `SoapySDR::log(SoapySDRLogLevel, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
CMakeFiles/LMS7Support.dir/Registration.cpp.o: In function `_GLOBAL__sub_I_Registration.cpp':
Registration.cpp:(.text.startup+0x68): undefined reference to `SoapySDR::Registry::Registry(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<st                                           d::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::b                                           asic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::                                           basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, std::allocator<std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::ba                                           sic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx1                                           1::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > > > (* const&)(std::map<std::__c                                           xx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<cha                                           r, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std [... cut ...]
Streaming.cpp:(.text+0x10cc): undefined reference to `SoapySDR::formatToSize(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
SoapyLMS7/CMakeFiles/LMS7Support.dir/build.make:150: recipe for target 'SoapyLMS7/libLMS7Support.so' failed
make[2]: *** [SoapyLMS7/libLMS7Support.so] Error 1
CMakeFiles/Makefile2:635: recipe for target 'SoapyLMS7/CMakeFiles/LMS7Support.dir/all' failed
make[1]: *** [SoapyLMS7/CMakeFiles/LMS7Support.dir/all] Error 2
Makefile:127: recipe for target 'all' failed

rjonaitis, 2018-05-02:
Looks like your SoapySDR library is not found. Ensure that you have it installed and CMake finds correct path.

IgnasJarusevicius, 2018-05-04:
You have a problem linking to SoapySDR library. libLMS7Support should not be build if SoapySDR is not detected by CMake. Maybe you have multiple versions of SoapySDR installed on your system?

aligrt, 2018-06-02:
resolved, First install SoapySDR and then LimeSuite, successfully build.

closed 2018-06-04 by rjonaitis