myriadrf/LimeSuite #190: Issue setting BW of 2nd channel with  18.04

darkstar007, 2018-05-13:
Hi, 

I get errors like the following:
```
Setting antenna to LNAH
[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 9060A02593414'
[INFO] Reference clock 30.72 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 3.072e+07 MHz
[INFO] LMS7002M calibration values caching Disable
Using device LimeSDR-USB: boardSerialNumber=0x9060a02593414 firmwareVersion=4 gatewareVersion=2.16 hardwareVersion=4 protocolVersion=1 
Found 6 antenna(s): NONE LNAH LNAL LNAW LB1 LB2 
Found 3 gain(s): TIA LNA PGA 
Found 2 frequencies: RF BB 
Found 2 sample rates: 100000 65000000 
Found 0 bandwidths: 
Channel info:    
Using output format: CS16 (input format CS16)
****Number of channels: 2
Sampling at 32000000 S/s on chan 0.
Tuned to 1156000000 Hz on channel 0.
****Antenna set to LNAH on chan 0
[INFO] RX LPF configured
Bandwidth parameter 28800000 Hz resulted in 28800000 Hz on chan 0.
Tuner gain set to 36.00 dB on chan 0.
Sampling at 32000000 S/s on chan 1.
Tuned to 1156000000 Hz on channel 1.
****Antenna set to LNAH on chan 1
[ERROR] Tune Rx Filter: MCU error 1 (Generic error)
[ERROR] setBandwidth(Rx, 1, 28.8 MHz) Failed - 
ERROR: Failed to set bandwidth on chan 1.
```

when trying the run my Lime with 2 channels. If I reduce the BW that both the channels use then it will work (I have to reduce it down to ~16MHz in this case before it will actually work - this is less than the bandwidth of my signal). Its always setting the BW of channel 2 that fails. I used to be able to use higher bandwidth in version 17 - so something has changed for the worse.

Thanks,

Matt

rjonaitis, 2018-06-04:
Can you do LMS_SaveConfig() after that failure and attach the file here?

darkstar007, 2018-06-05:
Unfortunately I won't have access to the Lime for the next 3 weeks.

darkstar007, 2018-06-26:
Attached is the file generated by SoapySDRDevice_writeSetting(dev, "SAVE_CONFIG", "lms_config.txt"); when SoapySDRDevice_setBandwidth(dev, SOAPY_SDR_RX, chan, (double)bandwidth) != 0.

[lms_config.txt](https://github.com/myriadrf/LimeSuite/files/2136019/lms_config.txt)

IgnasJarusevicius, 2018-06-26:
Do you still get the same error with latest LimeSuite 18.06? I tried loading your config file using LimeSutieGUI and Rx filter tuned just fine on both channels.

darkstar007, 2018-06-26:
Currently on 18.04 - I'll upgrade a machine to git master and try on that.

darkstar007, 2018-06-26:
Yes, 18.06 seems to work OK - and also seems to have fixed a number of (power?) issues I'd have been seeing at high sampling rates since around 17.0.

I'll leave this bug report open until the weekend just to give me some time to double check the data that gets recorded.

darkstar007, 2018-06-30:
I'll mark this as closed (in 18.06).

closed 2018-06-30 by darkstar007