myriadrf/LimeSuite #193: LimeSDR Mini: osmo-trx: Failure during LMS_Init()

pespin, 2018-05-25:
Using latest master LimeSuite:

```
Version information:
  Library version:      v18.04.2-g5caf4ce3
  Build timestamp:      2018-05-25
  Interface version:    v2018.3.0
  Binary interface:     18.04-1

Supported connections:
   * FT601
   * FX3
   * PCIEXillybus
```
osmo-trx fails to initialize a LimeSDR mini, but it works fine when using the bigger brother LimeSDR:

```
$ bin/osmo-trx-lms -C /home/pespin/dev/sysmocom/git/osmo-trx/doc/examples/osmo-trx-limesdr.cfg
Info: SSE3 support compiled in and supported by CPU
Info: SSE4.1 support compiled in and supported by CPU
Fri May 25 18:54:59 2018 DLGLOBAL <0002> telnet_interface.c:104 telnet at 127.0.0.1 4237
Fri May 25 18:54:59 2018 DLCTRL <0009> control_if.c:866 CTRL at 127.0.0.1 4236
Config Settings
   Log Level............... 1
   Device args.............
   TRX Base Port........... 5700
   TRX Address............. 192.168.30.100
   GSM BTS Address......... 192.168.30.1
   Channels................ 1
   Tx Samples-per-Symbol... 4
   Rx Samples-per-Symbol... 4
   EDGE support............ 0
   Reference............... 0
   C0 Filler Table......... 1
   Multi-Carrier........... 0
   Tuning offset........... 0
   RSSI to dBm offset...... 0
   Swap channels........... 0
   Tx Antennas............. '<default>'
   Rx Antennas............. '<default>'

Setting SCHED_RR priority(19)
Fri May 25 18:54:59 2018 DMAIN <0000> LMSDevice.cpp:46 [tid=139637053362048] creating LMS device...
Fri May 25 18:54:59 2018 DMAIN <0000> LMSDevice.cpp:88 [tid=139637053362048] Opening LMS device..
Fri May 25 18:54:59 2018 DMAIN <0000> LMSDevice.cpp:94 [tid=139637053362048] Devices found: 1
Fri May 25 18:54:59 2018 DMAIN <0000> LMSDevice.cpp:104 [tid=139637053362048] Device [0]: LimeSDR Mini, media=USB 2.0, module=FT601, addr=24607:1027, serial=1D3548A9613216
Fri May 25 18:54:59 2018 DLMS <0001> LMSDevice.cpp:71 [tid=139637053362048] Claimed Interface
Fri May 25 18:54:59 2018 DLMS <0001> LMSDevice.cpp:71 [tid=139637053362048] Estimated reference clock 40.0011 MHz
Fri May 25 18:54:59 2018 DLMS <0001> LMSDevice.cpp:71 [tid=139637053362048] Reference clock 40.00 MHz
Fri May 25 18:54:59 2018 DMAIN <0000> LMSDevice.cpp:115 [tid=139637053362048] Init LMS device
Fri May 25 18:54:59 2018 DLMS <0001> LMSDevice.cpp:71 [tid=139637053362048] INT 121, FRAC 0, DIV_LOCH 1, EN_DIV2_DIVPROG 0
Fri May 25 18:54:59 2018 DLMS <0001> LMSDevice.cpp:71 [tid=139637053362048] VCO 5000.00 MHz, RefClk 40.00 MHz
Fri May 25 18:54:59 2018 DLMS <0001> LMSDevice.cpp:71 [tid=139637053362048] ICT_VCO: 180
Fri May 25 18:54:59 2018 DLMS <0001> LMSDevice.cpp:71 [tid=139637053362048] csw=64      cmphl=0
Fri May 25 18:54:59 2018 DLMS <0001> LMSDevice.cpp:71 [tid=139637053362048] csw=96      cmphl=0
Fri May 25 18:54:59 2018 DLMS <0001> LMSDevice.cpp:71 [tid=139637053362048] csw=112     cmphl=0
Fri May 25 18:54:59 2018 DLMS <0001> LMSDevice.cpp:71 [tid=139637053362048] csw=120     cmphl=0
Fri May 25 18:54:59 2018 DLMS <0001> LMSDevice.cpp:71 [tid=139637053362048] csw=124     cmphl=0
Fri May 25 18:54:59 2018 DLMS <0001> LMSDevice.cpp:71 [tid=139637053362048] csw=126     cmphl=0
Fri May 25 18:54:59 2018 DLMS <0001> LMSDevice.cpp:71 [tid=139637053362048] csw=127     cmphl=0
Fri May 25 18:54:59 2018 DLMS <0001> LMSDevice.cpp:71 [tid=139637053362048] Failed to lock
Fri May 25 18:54:59 2018 DLMS <0001> LMSDevice.cpp:71 [tid=139637053362048] csw=192     cmphl=0
Fri May 25 18:54:59 2018 DLMS <0001> LMSDevice.cpp:71 [tid=139637053362048] csw=224     cmphl=2
Fri May 25 18:54:59 2018 DLMS <0001> LMSDevice.cpp:71 [tid=139637053362048] csw=240     cmphl=3
Fri May 25 18:54:59 2018 DLMS <0001> LMSDevice.cpp:71 [tid=139637053362048] csw=232     cmphl=3
Fri May 25 18:54:59 2018 DLMS <0001> LMSDevice.cpp:71 [tid=139637053362048] csw=228     cmphl=2
Fri May 25 18:54:59 2018 DLMS <0001> LMSDevice.cpp:71 [tid=139637053362048] csw=230     cmphl=2
Fri May 25 18:54:59 2018 DLMS <0001> LMSDevice.cpp:71 [tid=139637053362048] csw=231     cmphl=3
Fri May 25 18:54:59 2018 DLMS <0001> LMSDevice.cpp:71 [tid=139637053362048] Failed to lock
Fri May 25 18:54:59 2018 DLMS <0001> LMSDevice.cpp:71 [tid=139637053362048] cmphl=2
[... 104 lines cut ...]
            return -1;

        info_list = new lms_info_str_t[n];

        if (LMS_GetDeviceList(info_list) < 0)                //Populate device list
            LOG(ERROR) << "LMS_GetDeviceList(info_list) failed";

        for (i = 0; i < n; i++)                     //print device list
           LOG(DEBUG) << "Device [" << i << "]: " << info_list[i];

	rc = LMS_Open(&m_lms_dev, info_list[0], NULL);
	if (rc != 0) {
		LOG(ERROR) << "LMS_GetDeviceList() failed)";
		delete [] info_list;
		return -1;
	}

	delete [] info_list;

	LOG(INFO) << "Init LMS device";
	if (LMS_Init(m_lms_dev) != 0) {
		LOG(ERROR) << "LMS_Init() failed";
		return -1;
	}
```

ztamosevicius, 2018-05-28:
What version of LimeSDR-Mini board you are using? What GW version is installed?

pespin, 2018-05-28:
In my board I see printed: "LimeSDR Mini v1.0"

When I connect to it with LimeSuiteGUI, I get:
"INFO: Connected Control port: LimeSDR-Mini FW:5 HW:0 Protocol:1 GW:1.18 Ref Clk: 40.00 MHz"

USB 2 vs 3 should not matter at this time.

pespin, 2018-05-28:
I confirm the issue does NOT appear (it can be started) with a LimeSDR Mini v1.1, with following firmware information:
`INFO: Connected Control port: LimeSDR-Mini FW:5 HW:0 Protocol:1 GW:1.24 Ref Clk: 40.00 MHz`

9600, 2018-05-28:
The pre-production v1.0 board will be running quite old gateware that could only be updated via JTAG cable. Whether it's capable of being updated to a current gateware I'm not sure, but @ztamosevicius would be able to advise.

mariocannistra, 2018-05-29:
Hello @pespin , @9600 , @ztamosevicius .
I successfully reprogrammed the LimeSDR Mini using a 9 Euro clone of the Altera usb programmer.
You can find some posts with details and screenshots starting from this one in the forum: https://discourse.myriadrf.org/t/problems-with-limesdr-mini/2394/118 

There you will read several messages exchanged with @ztamosevicius on this topic and there is documentation on the JTAG cables at this wiki page:  https://wiki.myriadrf.org/LimeSDR-Mini_JTAG_cable_adapter#LimeSDR-Mini_JTAG_cable_adapter 

I used the Quartus sw to program the LimeSDR Mini board but I had to change the device driver of the clone usb programmer. The original one, installed with Quartus, does not work with the clone. So I used another one mentioned here: http://www.alteraforum.com/forum/showthread.php?t=19233

This is just a summary, please read all those posts and the wiki page before starting this process since there are lots of details involved.

mario

ztamosevicius, 2018-05-29:
LimeSDR-Mini v1.0 is not supported. Pre-production boards are v1.1, just gateware version is older than 24.

IgnasJarusevicius, 2018-06-22:
This seems to be answered. It was a problem with old preproduction board.

closed 2018-06-22 by IgnasJarusevicius