myriadrf/LimeSuite #194: LimeSDR Mini RPI3,  LimeUtil --info,  LimeUtil --find

oe5rnl, 2018-06-11:
Using latest master LimeSuite.

Only the PCIEXillybus connection is  available.

I need FT601 for the LimeMini - or not ?
Bug or is there a cmake switch ?

---------------------------------------------
LimeUtil --find  -> no device detected

---------------------------------------------
LimeUtil --info
######################################################
## LimeSuite information summary
######################################################

Version information:
  Library version:	v18.06.0-gf1e54444
  Build timestamp:	2018-06-11
  Interface version:	v2018.6.0
  Binary interface:	18.06-1

System resources:
  Installation root:	/usr/local
  User home directory:	/home/pi
  App data directory:	/home/pi/.local/share/LimeSuite
  Config directory:	/home/pi/.limesuite
  Image search paths:
     - /home/pi/.local/share/LimeSuite/images
     - /usr/local/share/LimeSuite/images

Supported connections:
   * PCIEXillybus
----------------------------------------------------------------------------

FFY00, 2018-06-11:
Did you install the udev rules?

oe5rnl, 2018-06-11:
Yes i did:
cd LimeSuite/udev-rules
chmod +x *sh
sudo ./install.sh

lsusb
Bus 001 Device 014: ID 0403:601f Future Technology Devices International, Ltd 

Info in /etc/udev/rules.d/64-limesuite.rules:

SUBSYSTEM=="usb", ATTR{idVendor}=="0403", ATTR{idProduct}=="601f", SYMLINK+="stream-%k", TAG+="uaccess"
---------------------------
This is my Hardware:
LimeSDR Mini v1.1
Firmware: 5
Hardware: 0
Protocol: 1
Gateway: 1.24
Gateway rev.: 1,24
Gateway Target: LimSDR-Mini

The LimeSDR-Mini works on Windows and Ubuntu 16.04

TNX

oe5rnl, 2018-06-11:
Could this the problem:

On the RPI3 (from latest repo) limeutil --info shows:

Binary interface:	18.06-1

On the Ubuntu 16.04  limeutil --info shows:

Binary interface:	18.04-1

The last update for LimeSDR-Mini on Ubuntu was:

LimeUtil --update
Connected to [LimeSDR Mini [USB 2.0] 1D3AD534B68430]
[100%] 303104/303104 Bytes programming: completed (/usr/share/LimeSuite/images/18.04/LimeSDR-Mini_HW_1.1_r1.24.rpd)
Programming update complete!

TNX

oe5rnl, 2018-06-12:
I found the solution :-)
---------
Raspberry PI3:

sudo apt-get install libusb-1.0-0-dev
---------
LimeUtil --info
######################################################
## LimeSuite information summary
######################################################

Version information:
  Library version:	v18.06.0-gf1e54444
  Build timestamp:	2018-06-12
  Interface version:	v2018.6.0
  Binary interface:	18.06-1

System resources:
  Installation root:	/usr/local
  User home directory:	/home/pi
  App data directory:	/home/pi/.local/share/LimeSuite
  Config directory:	/home/pi/.limesuite
  Image search paths:
     - /home/pi/.local/share/LimeSuite/images
     - /usr/local/share/LimeSuite/images

Supported connections:
   * FT601
   * FX3
   * PCIEXillybus
--------------------------

LimeUtil --find
  * [LimeSDR Mini, media=USB 2.0, module=FT601, addr=24607:1027, serial=1D3AD534B68430]
  * [LimeSDR-USB, media=USB 2.0, module=FX3, addr=1d50:6108, serial=0009062000C43419]
--------------------

i hope this help other folks.

TNX

closed 2018-06-12 by oe5rnl

ShitalBorganve, 2021-08-07:
Can i use RTL-SDR 
[ 8790.700908] rtl2832_sdr rtl2832_sdr.0.auto: Registered as swradio0
[ 8790.700921] rtl2832_sdr rtl2832_sdr.0.auto: Realtek RTL2832 SDR attached
[ 8790.700932] rtl2832_sdr rtl2832_sdr.0.auto: SDR API is still slightly experimental and functionality changes may follow
[ 8790.712909] Registered IR keymap rc-astrometa-t2hybrid
[ 8790.725165] IR NEC protocol handler initialized
In place of LimeSDR?
I am trying, but LimeUtil is unable to detect it.
Thank you.

9600, 2021-08-07:
@ShitalBorganve No. Lime Suite only supports LimeSDR/LimeNET hardware.

ShitalBorganve, 2021-08-08:
could you please suggest what other software stack i can use with RTL2832 SDR/HackRF one to achieve this?

9600, 2021-08-08:
Achieve what? This is an issue on the repo for a driver for a totally different SDR. Not an application. 

I'd suggest asking in the RTL and HackRF communities.

ShitalBorganve, 2021-08-08:
okay