myriadrf/LimeSuite #195: what's the min/max Tx gain? Is the API documentation wrong?

laf0rge, 2018-06-13:
From an application program point of view, when I support a large variety of devices, I want to programmatically discover the permitted minimum and maximum gain range for each gain that's configurable in any device.  At least the overall Rx and Tx gain should have functions to obtain the values.

However, there's only a LMS_SetGaindB function, and not a LMS_GetGaindBRange function, as there is for the band-width.

How is an application supposed to obtain the permitted gain ranges?

In the API documentation, it states `Desired gain, range [0, 70] for RX, [0, 60] for TX`.  So for transmit, I expect to be setting any value in the interval between [0..60 dB].  However, in reality it seems one can set much higher values.

Also, when seting the gain using the function to set it normalized, setting a normalized value of 1.0 and then calling LMS_GetGaindB() results in 76.

So why does the API documentation state that Tx Gain has a range up to 60, when it seem that values up to 76 appear to be legal?

IgnasJarusevicius, 2018-06-21:
Minimum Tx gain for LMS_SetGaindB() is 0 as it does not accept negative values. Maximum Tx gain is not strictly defined but range [0, 73] that is stated in LimeSuite.h seems like a reasonable guideline. Range interval [0, 60] is probably from outdated documentation (v17.12), 
You are right that normalized gain function with value of 1.0 attempts to set Tx gain to 76, but it does not always achieve that. 
When setting gain to higher values (65+), there may be a trade-off between output level and signal quality. So even if in some cases you can set Tx gain to 80+ it will probably be unusable.

IgnasJarusevicius, 2018-12-11:
I think the question has been answered. Closing.

closed 2018-12-11 by IgnasJarusevicius