myriadrf/LimeSuite #202: udev rules don't work on Raspbian.

9600, 2018-07-06:
Raspbian users have to run applications as root.

guruofquality, 2018-07-06:
I recently had an issue on a new machine that I setup. It was probably ubuntu 17.10, I will have to check.

I think its an issue of udev version/system config and using some features that may not be present. This is a current rule in limesuite: `SUBSYSTEM=="usb", ATTR{idVendor}=="1d50", ATTR{idProduct}=="6108", SYMLINK+="stream-%k", TAG+="uaccess"` I suspect the `TAG+="uaccess"` part

This is a rule from hackrf `ATTR{idVendor}=="1fc9", ATTR{idProduct}=="000c", SYMLINK+="nxp-dfu-%k", MODE="660", GROUP="plugdev"`

We might have to reverse this commit: https://github.com/myriadrf/LimeSuite/commit/b383b2a7f1349afa39e60a85b4bbd4ee259c92ae `MODE="660", GROUP="plugdev"` is probably better.

@andreasbombe any comment on the `TAG+="uaccess"` stuff for udev? I couldnt find good docs on it, but its not working out of the box on some debian/ubuntu systems. You wrote `The preferred way to make the device accessible to the logged in user is the uaccess tag.` Is there something to be configured for uaccess?

* Another reference https://wiki.archlinux.org/index.php/Talk:Udev#Use_of_.27uaccess.27_instead_of_GROUP_and_MODE.3F

csete, 2019-02-07:
Hi,

I too [noticed this last year](https://discourse.myriadrf.org/t/limesdr-permission-issues-on-debian-and-ubuntu-server/2391), though it seemed to me that it was more a server vs. desktop issue, i.e. the same problem exists on Ubuntu server, whereas it works fine on Ubuntu desktop (same version)

9600, 2019-02-08:
Right, so with Raspbian Lite I have to run applications as root, whereas on Raspbian Desktop the udev rules work. I also recently tried out Debian/aarch64 and had the same issue with the udev rules not working. Would be good to get this finally fixed.

IgnasJarusevicius, 2019-05-14:
I installed Raspbian Lite and did some investigation. The caveat with "uaccess" is that it gives permission to the user that is logged in locally. So it is not exactly server vs desktop issue, but user logged in locally vs user logged in only remotely. If you enable automatic console login on Raspbian Lite, it should give "pi" user access.
Looking at the reference provided by Josh, our old approach (using group and mode) is also not ideal as it relies on specific group existing and user beloning to that group.

IgnasJarusevicius, 2019-05-15:
@9600 @csete @guruofquality 
What are your thoughts on this? Should we revert limesuite udev rules to using "group and mode"? or keep "uaccess"? Or maybe just give access to all users with mode "666"? 
One of the main inconveniences with requiring root permission is for applications that could be launched from GUI and "uaccess" takes care of it. Other than that I am not sure how much this is system configuration versus user problem.

F5OEO, 2019-05-15:
For info, Edouard use https://github.com/f4exb/sdrangel/blob/master/udev-rules/64-limesuite.rules for SdrAngel and it works on Raspbian Lite and other systems.

9600, 2019-05-15:
Mode "666" does feel a bit insecure, but I guess it does the job.

closed 2019-05-15 by IgnasJarusevicius

csete, 2019-05-15:
I normally use group "plugdev" and mode 660, though there is no guarantee that users are automatically members of that group. So, mode 666 should work all the time.

commit dd83c404 (2019-05-15): Set MODE="666" in udev rules (resolves #202)