myriadrf/LimeSuite #207: LimeSDR Mini LED Indication

luigifreitas, 2018-07-18:
After closing the stream using the code below the LimeSDR Mini LED continues to blink as if streaming continued. This not happens with the LimeSDR USB, just the Mini version.

```
LMS_StopStream(&streamId);
LMS_DestroyStream(device, &streamId);
LMS_Close(device);
```

IgnasJarusevicius, 2018-07-23:
There is no stream activity indication on LimeSDR-Mini. The single LED on LimeSDR-Mini is for FPGA PLL lock indication and it behaves the same way as FPGA PLL lock LED on LimeSDR-USB.

luigifreitas, 2018-07-23:
I noticed at least two LED behaviors with the Mini.

State #1: (Red On, Green Blinking)
State #2: (Red Off, Green Blinking)

After I plug-in the Lime on the computer it goes to State #1. After I start some stream it goes to State #2. When I finish streaming it gets back to State #1.

I noticed that with LMS7 API it doesn’t get back to State #1 after closing the streaming. Is this an anomaly or the right state? (I could be wrong with this one but after the stream closes and the LED remains on State #1 the LimeSDR remains hot. I never noticed it getting hot with State #2.)

IgnasJarusevicius, 2018-07-23:
The LED blinking green (state 2) indicates that FPGA PLLs are locked. PLL lock state should not change after LMS_StopStream(), LMS_DestroyStream() or LMS_Close().

luigifreitas, 2018-07-23:
Understood. Closing this issue.

closed 2018-07-23 by luigifreitas