myriadrf/LimeSuite #208: Cannot build ocatve package

jscatena88, 2018-07-24:
first when running "make" inside of the octave folder after installing limesuite from the PPA originally I got the folowing error: 
`LimeSuite.cc:7:10: fatal error: LimeSuite.h: No such file or directory
 #include "LimeSuite.h"
`
So I then uninstalled from the ppa and used the build and install instructions for installing from source. After that I tried running make inside the octave folder again and got the following error:

`jason@jason-ThinkPad-X230:~/Worksapce/LimeSuite$ cd octave/
jason@jason-ThinkPad-X230:~/Worksapce/LimeSuite/octave$ make
mkoctfile -v -I/usr/local/include/lime -L. -lLimeSuite LimeSuite.cc
g++ -c -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/octave-4.2.2/octave/.. -I/usr/include/octave-4.2.2/octave -I/usr/include/hdf5/serial -I/usr/include/mpi  -pthread -fopenmp -g -O2 -fdebug-prefix-map=/build/octave-DtqyIg/octave-4.2.2=. -fstack-protector-strong -Wformat -Werror=format-security   -I/usr/local/include/lime  LimeSuite.cc -o LimeSuite.o
LimeSuite.cc: In function ‘void PrintDeviceInfo(lms_device_t*)’:
LimeSuite.cc:59:47: error: ‘const struct lms_dev_info_t’ has no member named ‘gatewareRevision’; did you mean ‘gatewareVersion’?
                       << " GW_rev: " << info->gatewareRevision << endl;
                                               ^~~~~~~~~~~~~~~~
                                               gatewareVersion
Makefile:21: recipe for target 'LimeSuite.oct' failed
make: *** [LimeSuite.oct] Error 1
jason@jason-ThinkPad-X230:~/Worksapce/LimeSuite/octave$ make
mkoctfile -v -I/usr/local/include/lime -L. -lLimeSuite LimeSuite.cc
g++ -c -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/octave-4.2.2/octave/.. -I/usr/include/octave-4.2.2/octave -I/usr/include/hdf5/serial -I/usr/include/mpi  -pthread -fopenmp -g -O2 -fdebug-prefix-map=/build/octave-DtqyIg/octave-4.2.2=. -fstack-protector-strong -Wformat -Werror=format-security   -I/usr/local/include/lime  LimeSuite.cc -o LimeSuite.o
LimeSuite.cc: In function ‘void PrintDeviceInfo(lms_device_t*)’:
LimeSuite.cc:59:47: error: ‘const struct lms_dev_info_t’ has no member named ‘gatewareRevision’; did you mean ‘gatewareVersion’?
                       << " GW_rev: " << info->gatewareRevision << endl;
                                               ^~~~~~~~~~~~~~~~
                                               gatewareVersion
Makefile:21: recipe for target 'LimeSuite.oct' failed
make: *** [LimeSuite.oct] Error 1
jason@jason-ThinkPad-X230:~/Worksapce/LimeSuite/octave$ 
`

guruofquality, 2018-07-24:
> first when running "make" inside of the octave folder after installing limesuite from the PPA originally I got the folowing error:
LimeSuite.cc:7:10: fatal error: LimeSuite.h: No such file or directory #include "LimeSuite.h"

just fyi for debian systems you need to install the development package liblimesuite-dev

rjonaitis, 2018-10-10:
makefile was using installed lime header, so there could have been mismatch between installed header and LimeSuite.cc source version. Anyway octave plugin compiling is now changed to use CMake https://github.com/myriadrf/LimeSuite/pull/226

closed 2018-10-10 by rjonaitis