myriadrf/LimeSuite #209: GFIR Coefficients Display appears faulty

brunaandrew, 2018-08-06:
Limesuite GUI GIR Filter to display filter coefficient is severely truncated horizontally and values are not visible (on Windows 10)

Also is there more information available on using GFIR?

Why is treatment of coefficients in GUI as 'float' whereas values are twos complement integer? 

In code for set and get GFIR coefficients, what is reason for '+ 24 * ...' as below
for (index = 0; index < coefCount; ++index)
        addresses.push_back(startAddr + index + 24 * (index / 40));

brunaandrew, 2018-08-06:
Tks for info - makes sense
I have been trying to load coefficient files and get the following behaviour irrespective of the number of coefficients (tested with 6 and 32)
RxTSP tab, only trying GFIR1
When the file 1st loaded - the dialog identifies the correct number of coefficients
On OK, the Length and Clk ratio values always automatically set their values to 1 irrespective of whether they were originally zero or say preset to 6 for length (for 32 coeff file)
[dump of register after each step shows the overwrite of previous values]
The 0x0480->onward register dump reveals the 1st two memory addresses in each memory bank have non-zero values - again irrespective of whether it is a 6 or 32 coeff file
In both cases the coeff table in the dialog is too narrow to see the values - although it is a bit wider for the 6 coeff file load

So overall there seem to be a few issues 
I am using version 18.04.1
Tks

brunaandrew, 2018-08-06:
oops didnt mean to close

closed 2018-08-06 by brunaandrew

IgnasJarusevicius, 2018-08-06:
This might help:
https://discourse.myriadrf.org/t/lms7002-digital-filtering-for-hf/564/40

brunaandrew, 2018-08-07:
Thanks - that's made a very big difference