myriadrf/LimeSuite #212: LMS7002M::GetTemperature() values dont make sense

guruofquality, 2018-08-23:
Tested as of writing with LimeSuite master on LimeSDR (and other boards). The temperature readout lingers around the range of lower to mid 50 degrees C. I applied a heat gun (100 C) to the LMS7 and verified the temperature with a FLIR and temperature probe to be well into 80 degrees C. LMS7 still reported mid 50 degrees C. I believe the scaling may be wrong.

gasparka, 2018-08-27:
I have confirmed this behavior on LimeSDR-Mini.

rjonaitis, 2018-11-02:
Fixed the possibility of infinite loop. But everything else seems to work as expected, applying hot air to the chip rose temperature to ~70 degrees. The coefficients were already adjusted 9 months ago, so I doubt that what you were experiencing was software problem.
LimeSDR-mini could show 51 degrees, if the SPI is not configured properly as then it would return 0 values for the calculations. As for the LimeSDR I don't know, it seems to work properly, @guruofquality could it be that your high temperature was affecting SPI communications?

guruofquality, 2018-11-03:
Thanks for the infinite loop fixed. I noticed it recently on some specific board(s), so it was unrelated to my temperature testing. Not sure as to why the loop didnt exit. it was not due to spi failure. :-/

I will try the test again on a few boards with heat gun and freeze spray and check that the spi bus is working while testing with a write/read test to an NCO value (or something). And report back.

gasparka, 2018-11-05:
I plugged in a cold LimeSDR-Mini and it reads 46C, while the board sensor reads 24C.
Something is not adding up...
[image]

guruofquality, 2018-11-06:
Ok I'm going to compare LimeSDR's temperature and the temperature on Iris which has 3 additional sensors. I will show them under ambient room temperature, doused with freeze spray, and lightly toasted with a heat gun.

**LimeSDR ambient**
```
temp is 48.812817 C
```

**LimeSDR heat gun**
```
temp is 53.058971 C
```

**LimeSDR freeze spray**
```
temp is 39.848717 C
```

**Iris ambient**
```
        : lms7 temp is 46.925640 C
        : zynq temp is 49.3393 C
        : rxfe temp is 43.812500 C
        : txfe temp is 25.250000 //note that tx PAs are off, rx is ON
```

**Iris heat gun**
```
        : lms7 temp is 50.700001 C
        : zynq temp is 74.9318 C
        : rxfe temp is 62.812500 C
        : txfe temp is 47.437500 C
```

**Iris freeze spray**
```
        : lms7 temp is 40.792305 C
        : zynq temp is 15.011 C
        : rxfe temp is 10.000000 C
        : txfe temp is -8.750000 C
```

I know this isnt a very exact test, and there are of course different temperature regions on the boards. But the temperature range that I'm seeing out of the other sensors for the same general application of heat and cold goes all the way from 0C to 70C. Whereas the LMS7 sees a range of maybe 38-53 degrees. I could try to push the LMS much more, but I'm afraid its just not reporting correctly, and I don't want to actually damage the board with heat, or waste tons of freeze spray. There has to be some kind of scaling issue.

btashton, 2018-11-06:
@rjonaitis in the discourse post I made I asked about how this block works because this code does not match what I was reading from the guide. It seems like a ratio should be being used.

davehawthorne, 2018-11-26:
I am seeing the same issue with incorrect temperature readings on my LimeSDR.

I am running LimeSuite 59d51d5 Release v18.10.0 and my hardware has the corresponding firmware.

Some further info from the debug trace in LMS7002M::GetTemperature():
[16:10:58] DEBUG: Vtemp 0x00A6, Vptat 0x00A5, Vdiff = -0.47, temp= 50.228
[16:10:58] DEBUG: Vtemp 0x00A7, Vptat 0x00A5, Vdiff = -0.94, temp= 49.756
[16:11:46] DEBUG: Vtemp 0x00A6, Vptat 0x00A3, Vdiff = -1.42, temp= 49.285
[16:31:29] DEBUG: Vtemp 0x00A6, Vptat 0x00A5, Vdiff = -0.47, temp= 50.228
[16:37:13] DEBUG: Vtemp 0x00A6, Vptat 0x00A5, Vdiff = -0.47, temp= 50.228
[16:52:39] DEBUG: Vtemp 0x00A6, Vptat 0x00A5, Vdiff = -0.47, temp= 50.228

It appears as though the MEASR_TPTAT_VAL value stays within the range 0xA3 to 0xA6.  Occasionally MEASR_TREF_VAL deviates from 0xA6 to 0xA7.

IgnasJarusevicius, 2019-05-13:
Hi,
I have adjusted  coefficients that are used for temperature calculations in ec615f429895c2bea7ce11e2e61eed0f10e0db5e
Now reported values should be much closer to actual temperatures. Check how it works with your boards.

closed 2019-05-24 by IgnasJarusevicius