myriadrf/LimeSuite #213: limesdr mini external clock command missing

rohlan, 2018-08-27:
when using osmo-trx-lms with a config using external 10MHz clock limesdr mini does now allow starting the trx:
```
Fri Aug 17 19:59:36 2018 DMAIN <0000> LMSDevice.cpp:142 [tid=140719660293952] Sample Rate: Host=1.08333e+06 RF=3.46667e+07
Fri Aug 17 19:59:36 2018 DMAIN <0000> LMSDevice.cpp:158 [tid=140719660293952] Setting External clock reference to 1e+07
Fri Aug 17 19:59:36 2018 DLMS <0001> LMSDevice.cpp:74 [tid=140719660293952] Command not supported
Fri Aug 17 19:59:36 2018 DMAIN <0000> LMSDevice.cpp:205 [tid=140719660293952] Error in LMS open, closing: Command not supported
Fri Aug 17 19:59:36 2018 DMAIN <0000> osmo-trx.cpp:453 [tid=140719660293952] Failed to create radio device
Shutting down transceiver...
```
this was tested with a limesdr mini 1.1 with 18.06 firm/gateware.

see https://osmocom.org/issues/3341 for more details

ztamosevicius, 2018-08-30:
You have to do a PCB mod to be able to use an external clock as described here:
https://wiki.myriadrf.org/LimeSDR-Mini_v1.1_hardware_description#Clock_Distribution

Then you have to change reference clock frequency in the software accordingly. 

Is a hardware mod done on your board?

IgnasJarusevicius, 2018-08-30:
Hi, 
I have made an update to LimeSuite (9977771c) so now setting external clock frequency for LimeSDR-Mini should just change reference clock instead of attempting to configure ADF4002 which is not present on LimeSDR-Mini. This should get rid of error when setting external clock for Mini.

IgnasJarusevicius, 2018-12-21:
Closing.
1. External clock command for LimeSDR-Mini has been added to LimeSuite.
2. Using external clock with LimeSDR-Mini requires hardware mod.
3. We have tested LimeSDR-Mini to work with 10MHz, 40MHz and 50MHz external clock.

closed 2018-12-21 by IgnasJarusevicius

rohlan, 2019-01-23:
how exactly is this supposed to be used?

i could not find any spec about the signal slope and levels so i started with a sinusoidal signal at 0dBm
some experimentation showed that the fpga(?) does only start up correctly when the signal level is >=3dBm.
in result i am doing my tests at 40MHz sinus at 4dBm.

i just updated a limesdr mini to gateware 28 and tested:
with 40MHz i seems to work, but LimeQuickTest fails on this section:
```
[ Clock Network Test ]
->REF clock test
  Test results: 1498; 14695; 27892 - PASSED
->VCTCXO test
  Results : 6711179 (min); 6711179 (max) - FAILED
  FAILED
->Clock Network Test FAILED
```
when supplying 10 or 50MHz i could not get the fpga to work (needs cold reset anyhow)

` LimeUtil --make`
 does not recognize the device at all, just outputs a (valid) serialno. all other values are 0 or UNKNOWN like when a bricked device is connected.

 `LimeQuickTest`
 or `osmo-trx-lms` do not start up/find the device when using different clocks than 40MHz.
i am recompiling osmo-trx-lms with a patch to use 40MHz against a LimeSuite HEAD for some real-world-application test now

rohlan, 2019-01-23:
osmo-trx-lms seems to work when patched, but i cannot get the phase errors small enough for validation yet.

rohlan, 2019-01-29:
ping? any news on how to use this and what is the intended waveform and levels would be extremely helpful.
i can not get the lime mini to work properly by supplying 10 or 50mhz.

ztamosevicius, 2019-02-01:
Check my answer here:
https://github.com/myriadrf/LimeSuite/issues/244#issuecomment-459658548