myriadrf/LimeSuite #216: LimeUtil --fpga requires hard resetting the device

gasparka, 2018-09-13:
`LimeUtil --fpga` works, but device is left in some sort of locked state i.e. i have to power-cycle to make it usable (LED is solid green after programming)
```bash
~> LimeUtil --fpga=/home/gaspar/git/LimeSDR-Mini_GW/LimeSDR-Mini_bitstreams/LimeSDR-Mini_lms7_trx_HW_1.2_auto.rpd 
Connected to [LimeSDR Mini [USB 3.0] 1D40EC49F23932]
Claimed Interface
FPGA configuring initiated
Programming finished, 303104 bytes sent! 18226 ms
FPGA configuring initiated

FPGA configuring initiated
FPGA boot from FLASH completed!
```

`LimeUtil --update` does not have this problem (red / yellow LED after programming):
```bash
~> LimeUtil --update
Connected to [LimeSDR Mini [USB 3.0] 1D40EC49F23932]
Claimed Interface
Command not supported
Programming finished, 303104 bytes sent! 17652 ms (/home/gaspar/.local/share/LimeSuite/images/18.06/LimeSDR-Mini_HW_1.1_r1.26.rpd)
FPGA configuring initiated

Programming update complete!
```

9600, 2018-09-28:
@ztamosevicius can we look into fixing this? We need to be able to update the gateware on remote systems without power cycling.

btashton, 2018-10-17:
@9600 @ztamosevicius What about consolidating the logic so that we just call `conn->ProgramUpdate` leveraging that code for both the gateware and firmware instead of repeating the logic in `LimeUtil.cpp`.  Happy to throw a patch together for that.

closed 2018-12-20 by IgnasJarusevicius

Mohammad-Yaprak, 2019-11-15:
Dear all We bought a LimeSDR USB. I started to connect and run by LimeSuiteGUI Unfortunately at the begining I got a probel in LED. two of the are ok. But the one related to FPGA is flashing red and other related to user defined is off. I tried several time several scenarios I install LimeSUite GUI in mac, I tried by USB 2 and 3. I tried by using power adaptor 6V 500 mA. But stil it has the problem. Please help me to solve this problem

commit 9f6d6ff7 (2018-12-20): LimeUtil: use LMS7_Device::Program() to flash GW/FW (resolves #216)