myriadrf/LimeSuite #220: NovenaRF7 linking errors

FFY00, 2018-09-30:
```
[ 39%] Linking CXX executable ../../bin/singleRX
/usr/bin/ld: ../libLimeSuite.so.18.10.0: undefined reference to `__loadConnectionNovenaRF7Entry()'
```

```
[ 40%] Linking CXX executable ../../bin/pll_sweep
/usr/bin/ld: ../libLimeSuite.so.18.10.0: undefined reference to `__loadConnectionNovenaRF7Entry()'
```

rjonaitis, 2018-10-01:
Duplicate of https://github.com/myriadrf/LimeSuite/issues/188
Novena code compilation was commented out, clear your cmake cache to remove old flags

FFY00, 2018-10-01:
I was enabling it. I am packaging LimeSuite for Arch Linux so I want to enable all possible features. Is there any plan to remove the code or fix it anytime soon?

IgnasJarusevicius, 2018-10-09:
Option ENABLE_NOVENARF7 should not even appear in CMake if you are using recent LimeSuite. Try clearing your build directory and CMake cache as rjonaitis suggested.

rjonaitis, 2018-10-09:
@IgnasJarusevicius is novena still being supported? As it makes no sense to have the code still present and the CMake option commented out. I've checked it is still compilable. As it was commented out in a merge commit without any reason, i suspect it was a mistake. So it should either be enabled again, or the code should be removed entirely.

IgnasJarusevicius, 2018-10-13:
ConnectionNovenaRF7 is not being supported and I am pretty sure that it would not work correctly with current version of LimeSuite (even if it compiles). I don't think there is a benefit in removing code from Git as it still might be useful for someone who has NovenaRF7 and it is easier to find this way.

rjonaitis, 2018-10-13:
Well, there is no mention anywhere that NovenaConnection was stopped being
supported and so having it in the current source does no good. As you just
mentioned even if the code is compiling  it is most probably not working,
which can mislead others that would try to use it (like this this case).
The best would be just to add a tag or mark branch where it was still in
the source (just in case anybody would still need to look it up) and from
that point remove it from master.

IgnasJarusevicius, 2018-12-11:
After some though I will leave it as it is (still in master but excluded from build). 
NovenaRF linking error should only appear in very rare cases when user has old build files from LimeSuite version that still had NovenaRF enabled. And with time such cases should completely disappear.
Separating it to different branch may be confusing as it kind of implies that there is working version for NovenaRF7 in that branch.

closed 2018-12-11 by IgnasJarusevicius