myriadrf/LimeSuite #228: Timestamp issue

lryindra, 2018-10-09:
Issue #219 was closed because SYNCH_DIS (bit 9) in DIQ interface control (0x0008) should always be 0 (Enable packets synchronization using timestamps). 

Because the FPGA must always use the time stamp, LMS_RecvStream and LMS_SendStream should return an error if meta is ever NULL because metadata.timestamp = 0 is not a valid condition. As found in src/API/lms7_api.cpp.

lryindra, 2018-10-10:
Given the following conditions:
<ul>
<li>SYNCH_DIS (bit 9) in DIQ interface control (0x0008) is 0 (Enable packets synchronization using timestamps)</li>
<li> .timestamp = 0 </li>
<li> .waitForTimestamp = false </li>
</ul>

I make a call to LMS_StopStream, then I make some changes to the data source, then I call LMS_StartStream.

What I observe is that data is only transmitted when I perform the LMS_StopStream & LMS_StartStream twice.

Issue #219 was the first workaround I found that made data transmit after every call to LMS_StartStream instead of every other.

IgnasJarusevicius, 2018-10-13:
Could you provide a more detailed explanation of your problem? What board are you using (LimeSDR-Mini or LimeSDR-USB)? Are you using Tx only or both Tx and Rx? 
I have tried replicating the problem by modifying basicTX example and I did not notice any problems with LimeSDR-USB or LimeSDR-Mini. 
The streaming sequence in my test was basically:
1. LMS_SetupStream()
2. LMS_StartStream()
3. Stream for 10s (using LMS_SendStream()) 
4. LMS_StopStream()
5. Sleep for some time (tried 100ms, 500ms and 1000ms)
6. LMS_StartStream()
7. Stream for 10s (using LMS_SendStream()) 
8. LMS_StopStream()
9. LMS_DestroyStream()

lryindra, 2018-10-15:
Hardware: Raspberry Pi 3 Model B
OS: Lubuntu 16.04 LTS from https://ubuntu-pi-flavour-maker.org/download/
attached software requires GnuRadio, LimeSuite (gitrev: ddae8b9), and gr-limesdr

[timestamp.tar.zip](https://github.com/myriadrf/LimeSuite/files/2480825/timestamp.tar.zip)

LimeSuite version 18.06 and 18.10 worked when the hardware was a Intel i5.  I haven't tried LimeSuite 18.10 on the Raspberry Pi.

lryindra, 2018-10-16:
I updated LimeSuite to gitrev: 18d2fa (this is LimeSuite 18.10), and I got the same results as I did with the older version of LimeSuite.

I should also say that I tested with a LimeSDR-Mini.

IgnasJarusevicius, 2018-10-17:
So the issue only appears on Raspberry Pi 3 and it works fine on PC with Intel i5 CPU?
RPi 3 only has USB2, did you use also USB2 on i5 PC? Are you using the latest FPGA gateware for LimeSDR-Mini? I think some USB2 issue was addressed in recent FPGA update for LimeSDR-Mini.
It is also possible that LimeSDR-Mini is not getting enough power from RPi via USB2 and is not working stably because of that.

lryindra, 2019-05-24:
I updated, so I'm posted the updated software:
Hardware: Raspberry Pi 3 Model B+
OS: ubuntu-18.04.2-preinstalled-server-arm64+raspi3.img.xz from https://www.ubuntu.com/download/iot/raspberry-pi-2-3
attached software required GnuRadio (version 3.7.13.5), LimeSuite (gui version 19.04.0-gdd83c404) and gr-limesdr (commit 4125e1f4f60a9c17698253c180911199bed657d1)
LimeSDR Mini [USB 2.0] FW:6 HW:1 Protocol:1 GW:1.30 Ref Clk: 40 MHz

[timestamp_test.tar.zip](https://github.com/myriadrf/LimeSuite/files/3218345/timestamp_test.tar.zip)

lryindra, 2019-05-24:
Do you have any idea how this issue will be resolved for the LimeSDR-PCIe?

lryindra, 2019-06-10:
https://discourse.myriadrf.org/t/rx-tx-timestamp-and-soapysdr/2459/8