myriadrf/LimeSuite #234: Missing runtime detection for SIMD flags

mnhauke, 2018-10-24:
It's nice to have some code that can be optimized optimzied via various SIMD instructions (AVX, SSE, ...) but from a distribution packagers point of view it's always inconvenient to choose this kind of optimizations at compile time.

Beeing the LimeSuite package maintainer for openSUSE I currently have the following choices:
 * build with SSE3 only so it will even run on older machines.
 * build with SSE4.2 or even AVX so it will work on "most modern" systems with a high risk of getting unhandled "illegal instruction" errors on hardware that do not support these SIMDs.
 * build several packages for LimeSuite, each with different SIMD flags enabled.

Binary Linux distributions like (open)SUSE, RedHat, Debian,... usually choose the option with maximum compatibility.

A better way to handle such optimizations would be to have a universal binary with a runtime detection for SIMD flags.

gasparka, 2018-10-25:
Not sure if LimeSuite even benefits from these optimizations, basically it is just passing around data.

rjonaitis, 2018-10-25:
Did not checked the disassembly, but the only place that could benefit from vectorization potentialy would be the samples unpacking from 12 bit to 16 bit values, as far as I remember there are no other functionality in the library that would get any significant benefit from these optimizations.
Furthermore, ability to choose the optimization at runtime would require splitting the library into several parts, compiled with different flags, and then dynamically loading appropriate one. That would only add unnecessary complexity to the library.
If lack of these optimizations would become a bottleneck in some project, I think just changing the flag and recompiling from source would be the least of their problem.

laf0rge, 2018-10-26:
FYI: In osmo-trx we had similar problems in the past until we added runtime CPU capability detection using __builtin_cpu_suports() as in examples seen at http://git.osmocom.org/osmo-trx/tree/Transceiver52M/arch/x86/convert.c

axilirator, 2018-10-26:
Please note that  __builtin_cpu_suports() is not a cross-platform way. But there is a library from Google: https://github.com/google/cpu_features.

IgnasJarusevicius, 2018-12-11:
Runtime detection of SIMD flags doesn't make much sense as there is no actual SIMD code in LimeSuite. As rjonaitis said, this would just add unnecessary complexity with little benefit.

Just package it without SSE flags or with SSE3,

closed 2018-12-11 by IgnasJarusevicius

f4exb, 2019-06-06:
The above linked issue in SDRangel tends to prove that there is some SIMD involved with the code may be incidentally but as detailed in the issue it can be limited to SSE3 so this should be fine.

f4exb, 2019-06-06:
In fact there are too many AVX instructions in LimeSuite code to be pasted here... just a sample:
```
737f4:       c5 db 59 05 64 c8 02    vmulsd 0x2c864(%rip),%xmm4,%xmm0        # a0060 <_ZTSN4lime8LMS7002ME+0x2f0>
```
I am using `objdump -d` and the `gawk` command (due to the size you have to use `gawk` in place of `awk`) as described here: https://stackoverflow.com/questions/47878352/how-to-check-if-compiled-code-uses-sse-and-avx-instructions