myriadrf/LimeSuite #237: LimeSDR Mini can't use 1.57542 GHz(TX)

SpenserCai, 2018-11-28:
```
LimeSuite Sink (TX) info

##################
Connecting to device
##################
Devices found: 1
##################
Device list:
Nr.:|0|device:|LimeSDR Mini, media=USB 3.0, module=FT601, addr=24607:1027, serial=1D423421DE7C61
##################
Device number 0 from the list is used.
Reference clock 40.00 MHz
##################

INFO: device_handler::set_chip_mode(): SISO mode set for device number 0.
```
**SetFrequencySXT(1575.42 MHz) - cannot deliver frequency**
```
INFO: device_handler::set_rf_freq(): RF frequency set [TX]: 1575.42 MHz.
INFO: device_handler::set_samp_rate(): set sampling rate: 10 MS/s.
Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
TX LPF configured
INFO: device_handler::set_analog_filter(): Configured bandwidth: 5 MHz.
GFIR LPF cannot be set to the requested bandwidth
INFO: device_handler::set_digital_filter(): digital filter channel 0 set [TX]: disabled.
INFO: device_handler::set_antenna(): channel 0 antenna set [TX]: BAND1.
INFO: device_handler::set_gain(): set gain [TX]: 30 dB.
INFO: device_handler::calibrate(): calibration channel 0 [TX]: disabled.
INFO: sink_impl::init_stream(): sink channel 0 (device nr. 0) stream setup done.
---------------------------------------------------------------
```

and then use LimeQuickTest
```
[ TESTING STARTED ]
->Start time: Wed Nov 28 12:11:37 2018

->Device: LimeSDR Mini, media=USB 3.0, module=FT601, addr=24607:1027, serial=1D423421DE7C61
  Serial Number: 1D423421DE7C61

[ Clock Network Test ]
->REF clock test
  Test results: 1708; 14905; 28102 - PASSED
->VCTCXO test
  Results : 6711093 (min); 6711245 (max) - PASSED
->Clock Network Test PASSED

[ FPGA EEPROM Test ]
->Read EEPROM
->Read data: 12 07 18 12 07 19 03
->FPGA EEPROM Test PASSED

[ LMS7002M Test ]
->Perform Registers Test
->External Reset line test
  Reg 0x20: Write value 0xFFFD, Read value 0xFFFD
  Reg 0x20: value after reset 0x0FFFF
->LMS7002M Test PASSED

[ RF Loopback Test ]
->Configure LMS
->Run Tests (TX_2 -> LNA_W):
  CH0 (SXR=1000.0MHz, SXT=1005.0MHz): Result:(-53.4 dBFS, -1.92 MHz) - FAILED
->Run Tests (TX_1 -> LNA_H):
  CH0 (SXR=2100.0MHz, SXT=2105.0MHz): Result:(-62.4 dBFS, -2.39 MHz) - FAILED
->RF Loopback Test FAILED

=> Board tests FAILED <=

Elapsed time: 2.78 seconds

```

if I reconnect LimeSDR will ok
```
[ TESTING STARTED ]
->Start time: Wed Nov 28 12:14:17 2018

->Device: LimeSDR Mini, media=USB 3.0, module=FT601, addr=24607:1027, serial=1D423421DE7C61
  Serial Number: 1D423421DE7C61

[ Clock Network Test ]
->REF clock test
  Test results: 3913; 17110; 30307 - PASSED
->VCTCXO test
  Results : 6711079 (min); 6711234 (max) - PASSED
->Clock Network Test PASSED

[ FPGA EEPROM Test ]
->Read EEPROM
->Read data: 12 07 18 12 07 19 03
->FPGA EEPROM Test PASSED

[ LMS7002M Test ]
->Perform Registers Test
->External Reset line test
  Reg 0x20: Write value 0xFFFD, Read value 0xFFFD
  Reg 0x20: value after reset 0x0FFFF
->LMS7002M Test PASSED

[ RF Loopback Test ]
->Configure LMS
->Run Tests (TX_2 -> LNA_W):
  CH0 (SXR=1000.0MHz, SXT=1005.0MHz): Result:(-13.5 dBFS, 5.00 MHz) - PASSED
->Run Tests (TX_1 -> LNA_H):
  CH0 (SXR=2100.0MHz, SXT=2105.0MHz): Result:(-14.4 dBFS, 5.00 MHz) - PASSED
->RF Loopback Test PASSED

=> Board tests PASSED <=

Elapsed time: 2.75 seconds

```

IgnasJarusevicius, 2018-12-11:
Is this specific to gr-limesdr? I tried tuning to that frequency using LimeSuiteGUI and it worked fine.

IgnasJarusevicius, 2019-02-26:
We cannot replicate this with any of our boards. It may be power or heat issue, or your hardware may be faulty. I can only suggest trying different PC or USB port that may be able to deliver more power or try cooling your LimeSDR-Mini with a fan (if you are not already doing that).
Closing as this is does not look like software issue.

closed 2019-02-26 by IgnasJarusevicius