myriadrf/LimeSuite #238: LMS_GetDeviceList leaks memory in osmo-trx-lms according to AddressSanitizer

pespin, 2018-11-29:
Using LimeSuite v18.10.0 (59d51d5) and osmo-trx-lms master (be8a83f681368307ca1c2d8726d27f5b98973231).

I usually run osmocom binaries with AddressSanitizer enabled (see --enable-sanitize configure option in osmocom projects). In case of osmo-trx, I build dependent libosmocore with AddressSanitizer but osmo-trx without it, since it was found that it may impact performance of critical paths which may interfere with proper operation.

In order to run osmo-trx-lms, I then need to preload libasan (since the main binary, osmo-trx doesn't use it):
LD_PRELOAD=/usr/lib/libasan.so.5.0.0 osmo-trx-lms -C /foo/bar/osmo-trx-lms.cfg

During startup of osmo-trx-lms, where LMS device is looked up and opened, I see LeakSanitizer (part of AddressSanitizer) printing a potential memory leak each time LMS_GetDeviceList() is used.

Here's the log output:

```
Thu Nov 29 18:19:55 2018 DDEV <0001> LMSDevice.cpp:49 [tid=140088174232064] creating LMS device...
Thu Nov 29 18:19:55 2018 DDEV <0001> LMSDevice.cpp:99 [tid=140088174232064] Opening LMS device..

=================================================================
==15656==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 12 byte(s) in 1 object(s) allocated from:
    #0 0x7f7d5e5a3019 in __interceptor_malloc /build/gcc/src/gcc/libsanitizer/asan/asan_malloc_linux.cc:86
    #1 0x5563ce47000e in xmalloc (/usr/bin/bash+0x7b00e)

SUMMARY: AddressSanitizer: 12 byte(s) leaked in 1 allocation(s).
Thu Nov 29 18:19:55 2018 DDEV <0001> LMSDevice.cpp:105 [tid=140088174232064] Devices found: 2

=================================================================
==15665==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 12 byte(s) in 1 object(s) allocated from:
    #0 0x7f86d75db019 in __interceptor_malloc /build/gcc/src/gcc/libsanitizer/asan/asan_malloc_linux.cc:86
    #1 0x558e3860200e in xmalloc (/usr/bin/bash+0x7b00e)

SUMMARY: AddressSanitizer: 12 byte(s) leaked in 1 allocation(s).
Thu Nov 29 18:19:56 2018 DDEV <0001> LMSDevice.cpp:115 [tid=140088174232064] Device [0]: LimeSDR-USB, media=USB 3.0, module=FX3, addr=1d50:6108, serial=0009060B00471827
Thu Nov 29 18:19:56 2018 DDEV <0001> LMSDevice.cpp:115 [tid=140088174232064] Device [1]: Remote, media=TCP, module=Z_Remote

=================================================================
==15679==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 12 byte(s) in 1 object(s) allocated from:
    #0 0x7f4c6968f019 in __interceptor_malloc /build/gcc/src/gcc/libsanitizer/asan/asan_malloc_linux.cc:86
    #1 0x55891880100e in xmalloc (/usr/bin/bash+0x7b00e)

SUMMARY: AddressSanitizer: 12 byte(s) leaked in 1 allocation(s).
Listening on port: 5000
Started listening thread
Thu Nov 29 18:24:22 2018 DLMS <0002> LMSDevice.cpp:74 [tid=140663147164160] Reference clock 30.72 MHz
Thu Nov 29 18:24:22 2018 DDEV <0001> LMSDevice.cpp:126 [tid=140663147164160] Init LMS device
```

And here's the related code in osmo-trx-lms:
https://git.osmocom.org/osmo-trx/tree/Transceiver52M/device/lms/LMSDevice.cpp?id=a439fed1662ce92c008e440b6a2ca1b2d95ce4cb#n89

So actually there's 3 leaks. First 2 account for the 2 times that LMS_GetDeviceList is called, and the 3rd one for the time LMS_Open is called. Interestingly the 3 of them leak same amount of memory, which means the leak is perhaps related in both places.

To confirm the issue seems to be in LMS_GetDeviceList, I did a few modifications to osmo-trx code and run it through gdb. I commented out our Log callback  (LMS_RegisterLogHandler(&lms_log_callback);) to make sure it's not used, and then added several calls like this in osmo-trx's LMSDevice::open:
n = LMS_GetDeviceList(NULL);
n = LMS_GetDeviceList(NULL);
n = LMS_GetDeviceList(NULL);

Then I add a breakpoint on LMS_GetDeviceList, and Indeed every time I press "continue" that goes from one LMS_GetDeviceList to the next, I get an AddressSanitizer memleak report.

pespin, 2018-11-29:
Is LimeSuite forking or creating a new process/shell when any of those APIs are called?

rjonaitis, 2018-11-30:
It does not fork or create processes. I've compiled simple loop, and checked with AddressSanitizer and with Valgrind, there are no signs of any leakage.
```C
for(int i=0; i<10000; ++i)
    int n = LMS_GetDeviceList(NULL);
```
**BUT** if AddressSanitizer is used with the LD_PRELOAD option, then it starts to pick up things outside the LimeSuite. Leaks like this get reported even when running just empty main() function.
```
==24230==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 96 byte(s) in 1 object(s) allocated from:
    #0 0x7fbef2310b90 in malloc (/usr/lib/x86_64-linux-gnu/libasan.so.5+0xedb90)
    #1 0x7fbef200d5aa in history_init (/usr/lib/x86_64-linux-gnu/libedit.so.2+0x225aa)

Indirect leak of 64 byte(s) in 1 object(s) allocated from:
    #0 0x7fbef2310b90 in malloc (/usr/lib/x86_64-linux-gnu/libasan.so.5+0xedb90)
    #1 0x7fbef200d54d  (/usr/lib/x86_64-linux-gnu/libedit.so.2+0x2254d)

SUMMARY: AddressSanitizer: 160 byte(s) leaked in 2 allocation(s).
```

closed 2018-12-05 by rjonaitis