myriadrf/LimeSuite #241: LimeSuite should build both Dynamic and Static libraries by default

racerxdl, 2019-01-03:
Currently its only possible to build Dynamic *or*  Static at a time. Ideally the default compilantion should generate both _.so_ and _.a_ files for both static linking and dynamic linking, keeping the choice of the target compiled program.

So far I can archieve that by running two compilations:

```bash
# Compile Static
cmake .. -DBUILD_SHARED_LIBS=OFF
make -j8
sudo make install

# Build Dynamic
cmake .. -DBUILD_SHARED_LIBS=ON
make -j8
sudo make install
```

That leaves both `libLimeSuite.a` and `libLimeSuite.so` at `/usr/lib/`.

PS: I will check if I can do that in CMakeLists, but looks like a lot of code is bound to `LIME_DLL` define.

IgnasJarusevicius, 2019-02-26:
I don't think this is an actual issue as there is an option to compile static library so you can compile it if necessary. Dynamic library is usually preferred so it is compiled by default. In my opinion pros/cons of compiling dynamic versus compiling both by default are debatable so it is not worth putting time into changing it.
Also, I have not investigated what the standard practice is, but taking a quick look at couple of projects that I have lying on my PC (uhd and wxwidgets) revealed that both of them also compile dynamic libraries only by default. 

In short, the current behaviour is not worth changing. Closing. Won't implement.

racerxdl, 2019-02-26:
Most of the linux LGPL libraries compiles both shared and static. Since LimeSuite is apache, it should provide both and let the developer decide at compile time of its program which one to use it. Even libstdc and glibc has dynamic and static. That also affects for example this: https://github.com/myriadrf/limedrv

Golang by default should use static versions of the library and cannot issue any build command as dep, this make a hassle when compiling a golang program.

9600, 2019-02-26:
@IgnasJarusevicius reopening. I think we need to consider this.

IgnasJarusevicius, 2019-02-28:
If it is necessary to build static library just (re)run library compilation with 'BUILD_SHARED_LIBS' set to OFF. It is one of common ways to do thing and is even documented in CMAKE https://cmake.org/cmake/help/v3.0/variable/BUILD_SHARED_LIBS.html 
So it is not like this is a bug or actual problem, just one of perfectly valid ways of doing things.

As far as SDRs goes, probably, the most used SDR library UHD behaves the same way in this regard. 

Also, I don't buy that most of the LGPL libraries compile both static and dynamic by default as my experience is otherwise (I am yet to encounter a library that builds both by default but I don't compile from source much) and it makes more sense to decide what is build by default based on common usage scenario and not on license. So it makes sense that some core library like libstdc would build both by default as its practically part of compiler and both are needed to function properly. On the other hand all of the GUI libraries that I have ever compiled from source (wxWidgets, Qt, FLTK) build only one by default as they probably see it as a common usage scenario (and I have checked that Qt and FLTK are listed as LGPL).

So why invest time in changing what is a valid way of doing things and dealing with potential caveats? There is option to compile static library so if you need static library just use the option and compile it.

rjonaitis, 2021-12-29:
Building both static and dynamic libs at the same time ain't going to happen. On windows that would result in filename conflicts, would need changing library names or directories. Also it would double compilation times when developing, which is undesirable when compiling on embedded systems. Basically such request adds too much problems for little benefit, which can already be achieved by just building library twice.

closed 2021-12-29 by rjonaitis