myriadrf/LimeSuite #244: limesdr mini external clock requirement documentation

rohlan, 2019-01-29:
i have done some experiments feeding external clock into limesdr mini.
so far it seems that its most happy when using 40MHz with fast slopes (as rectangle as possible)
sine works as well, but results in lots of phase noise (which is to be expected)
as for levels: i tried working with ~4-6dBm, but when using other clocks like 50Mhz or 10Mhz it needs higher signal levels like 13dBm to be accessible from limesuite.

some spec/documentation for the clock in and outputs would be nice.
reading the datasheet of LMK00105 and the schematic of limesdr mini i can gather that i need a rectangle clock with a risetime of 2V/ns or better for good phase noise performance. since its ac-coupled and has a 50ohm termination i think something in the range of ~1.5-3.3V ac would be good?

rohlan, 2019-01-31:
i have now fed a 10MHz rectangle clock to the limesdr mini. it got a amplitude of 2.5V.

sadly calls to LMS_Init() fail when using this.
as an experiment i duplicated the call to set the reference clock to be also _before_ LMS_Init() is called and could get osmo-trx to start again!

is there a documentation in which order functions of the limesuite api are supposed to be called?

also: is there any way to reset the limesdr mini without unplugging it from usb completely?
currently it needs a hard reset every time the clock is changed to another source or frequency.

rohlan, 2019-01-31:
screenshots of the clock signals last tested: https://osmocom.org/issues/3775

ztamosevicius, 2019-02-01:
External clock capabilities on LimeSDR-Mini board are defined by LMK00105 clock buffer specification. You must ensure voltage levels are in the range of LMK00105 capabilities. Ideally, we should use a phase detector circuitry to synchronize our TCXO to the external clock but there simply is no space on LimeSDR-Mini. Hence we decided just to provide an input to LMK clock buffer for external clock connection. But this means that user should pay attention on how external signal is connected, ensure proper voltage levels etc. 
It looks like you are supplying clock reference signal from signal generator. 4dBm is definitely too low. Set it to 10dBm. Note please, that LMK clock buffer expects positive voltage amplitude, while your generator may supply +/- voltage.
Regarding reset after clock frequency change. The proper way is to supply the reference clock first and then turn the board. This is the only way to have proper operation of the board. If frequency is changed on the fly, CPU inside of FPGA gets messed up hence no control over the board.

9600, 2019-02-01:
@ztamosevicius can we document this on the wiki?

ztamosevicius, 2019-02-01:
@9600 OK

ztamosevicius, 2019-05-13:
Clock distribution sections for LimeSDR-Mini 1v1 and 1v2 updated:
https://wiki.myriadrf.org/LimeSDR-Mini_v1.1_hardware_description#Clock_Distribution
https://wiki.myriadrf.org/LimeSDR-Mini_v1.2_hardware_description#Clock_Distribution

closed 2019-05-13 by ztamosevicius