myriadrf/LimeSuite #254: LimeSDR-Mini: no samples at second run on macOS

juribeparada, 2019-03-11:
When I run an application (first time) on macOS that communicates with LimeSDR-Mini, it works fine. But if I stop or close and then open again, LimeSDR-Mini does not transfer samples. This is reported here: https://discourse.myriadrf.org/t/limesdr-mini-macbook-pro-a1398-high-sierra-10-13-6-usb-hub-success/3326

Also, this issue produces trouble in CubicSDR on macOS, see this: https://github.com/cjcliffe/CubicSDR/pull/718

I have tried to investigate more this problem, and I found that is related with USB stream abort, specially when the driver calls libusb_cancel_transfer(). Looking at the implementation in libusb, I found that the issue is triggered when libusb calls the IOKit function ClearPipeStallBothEnds(). If I remove this line in libusb, the problem is "solved":

https://github.com/libusb/libusb/blob/2a7372db54094a406a755f0b8548b614ba8c78ec/libusb/os/darwin_usb.c#L1887

But I don't think doing this is a good idea. I hope this information could be useful to try to find a proper fix, if not, it would be nice any additional information to know where I have to look.

rjonaitis, 2019-04-13:
Seems to be fixed in libusb: https://github.com/libusb/libusb/issues/455

juribeparada, 2019-04-13:
Hi, that change in libusb doesn't fix this issue, but solved another issue that I reported here also: https://github.com/myriadrf/LimeSuite/issues/253 (and I closed when the reset fix was ready for libusb). This problem seems to me not related with libusb_reset_device(), but with libusb_cancel_transfer(), as explained before. Not sure if this problem must be solved in libusb or here, I only want to report it and see if any of the developers are aware of this problem and if have any idea how to investigate more deeply.