myriadrf/LimeSuite #255: Full reset without physically re-plugging

9600, 2019-03-11:
This is particularly important for remotely mounted SDRs, where it may not be possible to power cycle the host. We've seen a few requests for this, e.g.:

https://discourse.myriadrf.org/t/error-read-64-bytes-failed-and-cannot-deliver-frequency-after-receiving-for-some-time/3215/31

Gluttton, 2020-10-12:
Hello @9600 ,

LimeSDR API has already had a function:

    API_EXPORT int CALL_CONV LMS_Reset(lms_device_t *device);

According to the comments and implementation details, it seems that it provides the desired functionality. Doesn't it?

9600, 2020-10-12:
I couldn't say. 

@Zack, perhaps you could confirm?

zack, 2020-10-13:
negative, ghost rider. confirmation impossible.

@ztamosevicius, perhaps you could confirm?