myriadrf/LimeSuite #256: Enabling Frequency Divider checkbox in CLKGEN arbitrarily shifts between two states (shown on scope)

valer-cara, 2019-03-17:
Hi,

I don't fully understand what's going on under the hood, but I noticed that for lo-freq, the output when enabling the `VCO` in `CLKGEN` looks pretty noisy on the scope, whereas enabling `Frequency Divider` creates a clean carrier, but when enabling/disabling multiple times, it arbitrarily chooses two states (one lower amplitude and noisier, one cleaner and higher amplitude).

Behavior is shown in this video, test at 80mhz/12mhz in clkgen: https://www.youtube.com/watch?v=UdGe4u03gkA

Currently using:
[image]

IgnasJarusevicius, 2019-03-21:
Hello,
When you click on those checkboxes in LimeSuiteGUI, software only changes LMS chip register values. 
Can you provide more information about your test so I can forward it to hardware guys? What signal are you measuring on the scope?  Please save and provide your LMS chip register state to ini file so we can replicate your test. 

Also, I see that your board has old FW/GW so I advise you to update.

valer-cara, 2019-03-21:
Hi,

Thanks for the response! I'll update the FW and retry the test. I was probing the CH-A/TX-L, directly on the port (nothing else connected).

The test was pretty basic, here's how i reproduce it:
- start fresh, open the GUI, connect the device
- go to SXT, set freq to 96Mhz. Calculate & Tune.
- go to CLKGEN, set freq to 10Mhz. Calculate & Tune.
- on the CLKGEN tab, toggle `Power down controls -> Frequency Divider` multiple times and notice that when the checkbox is on, the output amplitude arbitrarily lands on one of two values. (in my case, either 4V or 7V peak to peak on the scope if i'm reading this correctly)

Again, I'm pretty clueless on what this means, but I thought this behavior is pretty strange. Perhaps it's the way I'm measuring that doesn't make sense. Anyway, hope it helps.

Thanks!

valer-cara, 2019-03-21:
Updated the FW, similar. I've attached the INI.

[test-freq-divider-ini.txt](https://github.com/myriadrf/LimeSuite/files/2994319/test-freq-divider-ini.txt)

IgnasJarusevicius, 2019-05-10:
Sorry for long delay. I do not have an explanation for what you see. We do not understand what are  you trying to achieve by toggling frequency divider. Your chip state does not seem to be configured for operation so it doesn't look like the issue is related to some failure during transmitting/receiving.  We are unable to see the issue here, sorry. Closing.

closed 2019-05-10 by IgnasJarusevicius