myriadrf/LimeSuite #260: bestVCO and bestCSW uninitialized

fmhess, 2019-04-07:
I'm building the MCU code with sdcc, and it found some initialization problems with 2 variables:

sdcc --std-c99 -mmcs51 -I ../../mcu_program//mcu_src/ -I ../../mcu_program/common_src/ -c ../../mcu_program/common_src/lms7002m_controls.c
../../mcu_program/common_src/lms7002m_controls.c:277: warning 84: 'auto' variable 'bestVCO' may be used before initialization
../../mcu_program/common_src/lms7002m_controls.c:278: warning 84: 'auto' variable 'bestCSW' may be used before initialization

I looked at the code, the warning is valid

rjonaitis, 2019-04-07:
true, that warning is valid, but the only case when the uninitialized path could be taken is if all of the chip's VCOs fail to tune, so in that case it doesn't matter which one is selected in the end. Those variables were left uninitialized to save program memory (as of now there is only 110 free bytes left of the 16KB of MCU memory). I guess now two bytes could be spared to initialize them, although it's not necessary.

Pay attention to the compiled program size when using SDCC as it might produce program image larger than 16KB, then it would not fit inside the MCU memory and won't work.

fmhess, 2019-04-07:
More use of uninitialized variables (these caused by goto).  This output is after I replaced all usage of the "pow" function with "powf".  "pow" uses type double and is not supported by sdcc.  There was a similar problem with log10/log10f in lms7002m_agc.c

```
sdcc --std-c99 -mmcs51 -I ../../mcu_program//mcu_src/ -I ../../mcu_program/common_src/ -c ../../mcu_program/common_src/lms7002m_filters.c
../../mcu_program/common_src/lms7002m_filters.c:284: warning 185: comparison of 'signed char' with 'unsigned char' requires promotion to int
../../mcu_program/common_src/lms7002m_filters.c:287: warning 185: comparison of 'signed char' with 'unsigned char' requires promotion to int
../../mcu_program/common_src/lms7002m_filters.c:289: warning 185: comparison of 'signed char' with 'unsigned char' requires promotion to int
../../mcu_program/common_src/lms7002m_filters.c:293: warning 185: comparison of 'signed char' with 'unsigned char' requires promotion to int
../../mcu_program/common_src/lms7002m_filters.c:420: warning 84: 'auto' variable 'ccomp_cfb_tia_rfe' may be used before initialization
../../mcu_program/common_src/lms7002m_filters.c:421: warning 84: 'auto' variable 'rcc_c_ctl_lpfl_rbb' may be used before initialization
../../mcu_program/common_src/lms7002m_filters.c:422: warning 84: 'auto' variable 'rcc_c_ctl_pga_rbb' may be used before initialization
../../mcu_program/common_src/lms7002m_filters.c:423: warning 84: 'auto' variable 'rcc_c_ctl_lpfh_rbb' may be used before initialization
../../mcu_program/common_src/lms7002m_filters.c:424: warning 84: 'auto' variable 'input_ctl_pga_rbb' may be used before initialization
../../mcu_program/common_src/lms7002m_filters.c:425: warning 84: 'auto' variable 'rcomp_tia_rfe' may be used before initialization
../../mcu_program/common_src/lms7002m_filters.c:427: warning 84: 'auto' variable 'pd_lpfhl' may be used before initialization
../../mcu_program/common_src/lms7002m_filters.c:675: warning 84: 'auto' variable 'ccal_lpflad_tbb' may be used before initialization
../../mcu_program/common_src/lms7002m_filters.c:678: warning 84: 'auto' variable 'rcal_lpfh_lpflad_tbb' may be used before initialization
```

rjonaitis, 2019-04-07:
Just to save you some time, if your goal is just to compile the existing MCU source code using SDCC, it is possible, but it will definetely won't fix into 16KB, I've tried that long time ago. You would have to either throw out some functionality or try to rewrite the code specifically for SDCC optimizations and hope it fits.

fmhess, 2019-04-08:
Yes, that is what I was doing.  sdcc wanted about 32K for the code, and also had problems running out of room in the internal 128 bytes memory for the DSEG.  Maybe I'll try compiling and running it on the host side.

rjonaitis, 2019-04-13:
Closing this issue, as the uninitialized variables and goto usage is intentional.

closed 2019-04-13 by rjonaitis