myriadrf/LimeSuite #261: SoapySDR Interface causing segfault for rx stream

nunnsy, 2019-04-12:
I've been trying to develop an application using the Python SoapySDR interface for a LimeSDR. When streaming samples over time, after some time (it varies), the Python interpreter crashes with a segfault.

Below is a snippet of the code running. Then the output in my terminal.

```
def rx_chain():
    global big_boi
    sdr = limey.get_sdr()
    rxs = limey.get_rx_stream()
    while True:
        print("#0")
        print("SDR: " + str(sdr))
        print("RXS: " + str(rxs))
        print("ea0: " + str(hash(ea0.data.tobytes())))
        print("ea1: " + str(hash(ea1.data.tobytes())))
        sr = sdr.readStream(rxs, [ea0, ea1], len(ea0))
        # power, psd_freq = plt.psd(ea0, NFFT=1024, Fs=limey.get_sample_rate()["rx0"]/1e6, Fc=2.4e9/1e6)
        print("#1")
        print(sr)
        if (sr.ret > 0):
            big_boi = np.concatenate((ea0[0:sr.ret], big_boi[0:-sr.ret]))
            print("#2")
            plt.cla()
            print("#3")
            plt.specgram(big_boi, NFFT=samples, Fs=f_s, Fc=431e6)
            print("#4")
            plt.pause(0.05)
        time.sleep(1)
```

```
ret=512, flags=4, timeNs=92993349000
#2
#3
#4
#0
SDR: FX3:LimeSDR-USB
RXS: <Swig Object of type 'SoapySDR::Stream *' at 0x7fc12c268930>
ea0: -8597006528890416220
ea1: -8055776645099117789
#1
ret=512, flags=4, timeNs=94133301000
#2
#3
#4
#0
SDR: FX3:LimeSDR-USB
RXS: <Swig Object of type 'SoapySDR::Stream *' at 0x7fc12c268930>
ea0: 931463062287685360
ea1: 2884037698141062820
Segmentation fault (core dumped)
###@ubuntu:~###/$ 
```

Are there any further things I can look at within the Python environment to better narrow this down? It appears all objects being used for the `.readStream()` function seem valid. This error occurs (usually) after 10 loops, but can last for over 60 before an error appears.

I'm assuming the issue will be contained within: https://github.com/myriadrf/LimeSuite/blob/master/SoapyLMS7/Streaming.cpp

Any help would be greatly appreciated!

rjonaitis, 2019-04-13:
The best you can do is post the whole python source, or at least minimal amount of code that could be executed to reproduce such crash. Otherwise it's pointless to speculate what could be happening without knowing what your program actually does.

nunnsy, 2019-04-14:
Not a problem, @rjonaitis - please excuse the messy code, it's been highly dynamic in trying to get initial tests to work smoothly.

The file which you saw run is:
```
import sys
sys.path.append("..") # Adds higher directory to python modules path.

import array, os
import numpy as np
from scipy import fftpack
import matplotlib.pyplot as plt
import matplotlib
import math
import time
import threading

matplotlib.use('Qt5Agg')

from lib.pySnowcone.sdr.limesdr import LimeSdr

samples = 512

limey = LimeSdr(431e6, 10e6)

# fa = os.open("./fifoA", os.O_WRONLY|os.O_CREAT)
# fb = os.open("./fifoB", os.O_WRONLY|os.O_CREAT)

ea0 = np.array([1]*samples, np.complex64)
ea1 = np.array([1]*samples, np.complex64)

big_boi = np.array([0]*samples*100, np.complex64)

f_s = limey.get_sample_rate()["rx0"]

def tx_carrier():
    wave_freq = 1e6
    rate = 10000000
    tx_stream = limey.get_tx_stream()
    sdr = limey.get_sdr()
    ampl = 0.8

    phase_acc = 0
    phase_inc = 2*math.pi*wave_freq/rate
    stream_mtu = sdr.getStreamMTU(tx_stream)
    samps_chan = np.array([ampl]*stream_mtu, np.complex64)

    time_last_print = time.time()
    total_samps = 0

    while True:
        phase_acc_next = phase_acc + stream_mtu*phase_inc
        phases = np.linspace(phase_acc, phase_acc_next, stream_mtu)
        samps_chan = ampl*np.exp(1j * phases).astype(np.complex64)
        phase_acc = phase_acc_next
        while phase_acc > math.pi * 2:
            phase_acc -= math.pi * 2

        print("Write 1")
        status = sdr.writeStream(tx_stream, [samps_chan], samps_chan.size, timeoutUs=100000000)
        print("Write 2")
        if status.ret != samps_chan.size:
            print(status.ret)
            # raise Exception("Expected writeStream() to consume all samples! %d" % status.ret)
            print("Expected writeStream() to consume all samples! %d" % status.ret)
        else:
            total_samps += status.ret

        if time.time() > time_last_print + 5.0:
            rate = total_samps / (time.time() - time_last_print) / 1e6
            print("Python siggen rate: %f Msps" % rate)
[... 247 lines cut ...]
        #         'correlation(%d) 0 does not match argmax(%d), probably bad data' %
        #         (rx0_coor_index, rx0_argmax_index))

        # if abs(rx1_coor_index-rx1_argmax_index) > len(tx_pulse_norm)/4:
        #     raise Exception(
        #         'correlation(%d) 1 does not match argmax(%d), probably bad data' %
        #         (rx1_coor_index, rx1_argmax_index))

        #calculate time offset
        tx_peak_time = int(tx_time_0 + (tx_argmax_index / rate) * 1e9)
        rx0_peak_time = int(rx_time_0 + (rx0_argmax_index / rate) * 1e9)
        rx1_peak_time = int(rx_time_0 + (rx1_argmax_index / rate) * 1e9)
        time_delta_0 = rx0_peak_time - tx_peak_time
        time_delta_1 = rx1_peak_time - tx_peak_time
        print('>>> Time delta %f us'%((time_delta_1 - time_delta_0) / 1e3))
        self._sample_offset = rx1_argmax_index - rx0_argmax_index
        print('>>> Sample Delta %d'%(self._sample_offset))
        print("Done!")
```

With my so far limited experience with SoapySDR, I figured the best way to synchronise two channels was following examples by @guruofquality : https://github.com/pothosware/SoapySDR/tree/master/python/apps

I figured, once a delay is measured, those streams should remain open. I understand that the buffer will fill in the mean time, and I'll need to empty it out again before the synchronised streaming occurs, but using the same streams should, after the buffer empty, have roughly the same sample delay?

If I am going about this completely wrong, please let me know and I'll pursue another avenue. I am tempted to move over to C/C++ due to the large amounts of streaming data I want to work with in the future. If this helps sort out streaming problems now, that would be even more reason to do so.

IgnasJarusevicius, 2019-05-15:
Hi,
I ran your code and found one place where SoapyLMS7 segfaults in Rx. For me, it now longer crashes after 8ee7053c1043262dbc78730a53525398e50c2355. Could you confirm that it fixes your problem?

nunnsy, 2019-05-16:
Hi @IgnasJarusevicius,

I will test it as soon as I have the opportunity (next few days). Thank you for alerting me to this update!

nunnsy, 2019-05-26:
Seems to have fixed the stability! Awesome, thank you for the assistance. 😄

closed 2019-05-26 by nunnsy