myriadrf/LimeSuite #262: signal gaps at 10 Mhz sampling rate

DennisCh1989, 2019-04-12:
I sent chirp at sampling rate 10 Mhz and saw gaps. it means that lime sdr mini add zeros on tx or rx .  how can i solve the problem ?

DennisCh1989, 2019-04-12:
after a gap chirp starts at the previous instantaneous frequency...

DennisCh1989, 2019-04-12:
i use only lemesuite not soapy sdr

rjonaitis, 2019-04-13:
It depends on how your samples transmission code works. Gaps could appear if your code does not provide samples fast enough. Do you use timestamps synchronization?

DennisCh1989, 2019-04-18:
thanks for your replay , but i only passed signal(chirp) from the file and got gaps

IgnasJarusevicius, 2019-05-10:
Can you post the simplest code that reproduces the issue? Also what OS are you using? Is LimeSDR connected via USB2 or USB3?
From your description It sound exactly like samples are not being send to LimeSDR in time. Something like: 1) send some samples (start transmitting signal), pause (gap), send some more samples (signal continues from last point before gap).

IgnasJarusevicius, 2019-05-24:
I do not observe the issue using basicTX example (with 10 MHz sample rate). 
There has been bug that could cause higher than necessary system load at higher sample rates that is now fixed 8efa0c5a94e5b1148ba9f88074bc90f30a0ecadd. The fix could help on lower performance systems. However, it does not make a difference for basicTX on my ~6-7 year old i7. 

Unable to reproduce. May be system performance problem.

closed 2019-05-24 by IgnasJarusevicius