myriadrf/LimeSuite #263: osmo-trx-lms + limesuite with real time priority makes kernel eat all available memory in system

pespin, 2019-05-03:
Please see https://osmocom.org/issues/3971 for a detailed description. It's fine to keep discussion in this github report if you prefer.

IgnasJarusevicius, 2019-05-10:
Hi,
I cannot think of the place in LimeSuite that could make kernel eat memory. Do you have an approximate idea at what point it happens (e.g. when opening device, during configuration, after starting sample stream)?
My first guess would be that extensive logging (writing to console, file, socket) somewhere is causing this. Have you tried changing logging level? 
LimeSuite itself should not use select(), poll() or accept() calls. But these call are normally used with sockets, so maybe that could give some hint.

pespin, 2019-05-10:
Hi,
I still need to run linux perf on it to try to find some code paths where this happens. As far as I remember it's enough to open the device and no need to call readSamples or writeSamples or similar. I'm pretty sure the mem alloc is somehow related to libusb and USB handling in the kernel. But i don't see this kind of issue when running other devices using UHD with same rt-prio. I don't see any extensive logging, and if it was it would most probably show up as memory allocated by osmo-trx process, which is not the case. Furthermore, I don't see huge amount of related syscalls (open, read, write). IIRC I saw lots of accept()/poll() calls coming from libusb threads.

IgnasJarusevicius, 2019-05-13:
If it libusb ralated, then it might be "libusb_handle_events_timeout_completed()" that is being run in a loop. The timeout is set to 250ms in LimeSuite so it blocks up to 250ms if there is no USB events. I don't know what it does internally while waiting for events.

pespin, 2019-08-02:
Quick update on this one:
* I deleted original osmocom redmine ticket by error, so link shared above is no longer valid. I recreated the ticket here: https://osmocom.org/issues/4136

* According to linux perf, mass filp allocations seem to come from accept() syscall, which ends up calling `__alloc_file()` in kernel: `kmem_cache_zalloc(filp_cachep, GFP_KERNEL);`

* Only accept() call I could find in LimeSuite is in src/ConnectionRemote/LMS64CProtocol_remote.cpp:119: `clientFd = accept(socketFd, (struct sockaddr*)&cli_addr, &clilen);`

So it's most probably related to libusb?

BTW, I can still trigger the bug by setting sched_setscheduler(SCHED_RR) with param.sched_priority= 18 in osmo-trx-lms, using following versions:
LimeSite: 19.04.0.r0.abf82140
kernel: 5.2.2-arch1-1-ARCH x86_64
libusb: 1.0.22-1

pespin, 2019-08-02:
So after further debugging, it is definetly a bug in (or at least triggered by) LimeSuite. The problem is basically socketFd being set as non-blocking and having a thread doing "while(1) rc = accept(socketFd)", where rc is all the time -1 and errnor = EAGAIN. That thread with RT_PRIO 18 seems to be causing trouble in the kernel. For some unknown reason, the kernel allocates a new client socket upon each accept() call, and probably schedules a deferred free, but with such a huge priority, it seems allocations happen much quicker than frees....

See https://osmocom.org/issues/4136 for more information.

So the solution to this issue is basically to stop hammering accept() continuously.  Use select() there and check for socketFd having read flag set, then and only then call accept() (it's fine to leave the socket non-blocking). Since you may want to exit the thread, add some other fd you can signal (like a pipe or a timerfd (man timerfd_create), and if select returns that fd is flagged, then exit the loop.

IgnasJarusevicius, 2019-08-29:
I can only see accept() in one place in LimeSutie and that code is disabled in default build as it is meant to be used for testing/debugging.

pespin, 2019-08-29:
Yes, it's the accept() enabled by that flag. Archlinux packages happen to enable it:
https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=limesuite-git

Details can be found https://osmocom.org/issues/4136#note-4 and onward

commit cfc53ed6 (2022-09-08): Disable RemoteConnection option as it's only for internal debugging only (Fix #263)