myriadrf/LimeSuite #265: LimeSuite: Improve documentation for lms_stream_status_t

pespin, 2019-06-04:
It is currently not clear when reading the documentation what fields of lms_stream_status_t mean (based on Rx/Tx direction differences) and how they should be handled.

For instance, I first thought droppedPacket, underrun, overrun values were "ever increasing" values over time. Then, after experimenting a bit I saw that actually those values were reset, and I initially though they were reset upon LMS_{Recv,Sends}Stream calls. But after looking at the code it seems they are actually reset upon each call to LMS_GetStreamStatus? Please update LMS_GetStreamStatus() LimeSuite.h doc to explain how/which fields are reset.

Also, please describe what does an "Rx overrun", "Rx underrun", "Tx overrun", "Tx underrun" (and same for other droppedPacket and other fields if apply) in the lms_stream_status_t section.

laf0rge, 2019-06-04:
Everone knows of course what an underrun/overrun is in principle.  But the important information is what do they specifically mean in the context of LimeSDR / LimeSuite.  It is really important to understand the full picture here when debugging issues.

IgnasJarusevicius, 2019-07-29:
Doxygen commnets has been updated.
https://github.com/myriadrf/LimeSuite/blob/master/src/lime/LimeSuite.h#L1094

In the latest master fifo underrun is counted when Tx is running but FIFO is empty for >100 ms (500ms in older versions).
In the latest master fifo overrun is counted when Rx FIFO is full but new data comes from the board and oldest samples in FIFO are overwritten (always 0 in previous versions).
Dropped packets in Tx are counted only when timestamps are enabled and SDR drops packet because of late timestamp.
Dropped packets in Rx are counted when gaps in Rx timestamps are detected (likely because buffer oveflow in hardware)

closed 2019-07-29 by IgnasJarusevicius