myriadrf/LimeSuite #266: Minimum TX sample rate unnecessarily limited?

sw, 2019-06-21:
The following casts to uint8_t limits the minimal TX sample rate to about 45kHz:
https://github.com/myriadrf/LimeSuite/blob/0d5027fb1ec582970bcdcb5cb19b4ce430cfbd12/src/lms7002m/LMS7002M.cpp#L1129

Is this really necessary? Changing the types of the three iHdiv* variables to uint16_t works for me with lower sample rates.

rjonaitis, 2019-06-21:
That parameter is only 8 bits ( 0x0089[10:3] ) in hardware, so it can't have larger values.
https://github.com/myriadrf/LimeSuite/blob/01e2d00c5005b85d1f94cca02881756a72e35e2a/src/lime/LMS7002M_parameters.h#L179

sw, 2019-06-21:
I wanted to use 38.4kHz -> freq_Hz = 4.9152MHz
```c
uint8_t iHdiv_high = (2.94e9/2 / freq_Hz)-1; // = 289 --> overflow, wraps around to 33
```
But shouldn't it be possible to do:
```c
uint16_t iHdiv_high = (2.94e9/2 / freq_Hz)-1; // = 289
uint8_t iHdiv_low = (1.93e9/2 / freq_Hz); // = 196
uint8_t iHdiv = (iHdiv_low + iHdiv_high)/2; // = 242
```

closed 2019-06-25 by IgnasJarusevicius