myriadrf/LimeSuite #268: Add "flags" field to lms_stream is not backward compatible

F5OEO, 2019-07-17:
With commit https://github.com/myriadrf/LimeSuite/commit/570e879ef5b04b73ab284016160ec87b6b668b14 which add a field to lms_stream could break some third software compilation.

Please announce it !

IgnasJarusevicius, 2019-07-18:
First of all, it is master branch where experimental features and (potential) API breakages occur. The cahnge may or not make it to next release or stable branch. The feature itself should get into next release but flags may be combined into channel field (which internally is 8bits only so other 24bit are currently wasted). I would prefer separate field as it is clearer/more readable but it needs to be evaluated how much problems such change can cause. 
Secondly, where do you get compilations errors? All old fields are still there so there should be no compilation errors for code that was not aware of the new field (am I missing something?).

F5OEO, 2019-07-18:
Agree Ignas, master branch is not the release one. I maybe hurry to use latest commits ;) About compatibility with third party software : maybe an ugly C-Style from my side in a C++. See https://github.com/F5OEO/limesdr_toolbox/commit/43158c0293fee174e559f62319c59486b1eae70e
Error from the compiler : limesdr_dvb.cpp:643:2: sorry, unimplemented: non-trivial designated initializers not supported.
This issue was a Warn for you for a positive criticism...not a complain.

IgnasJarusevicius, 2019-07-18:
Thanks for warning about possible problem with C style constructs in C++ code, I was not aware of that. 
There is nothing wrong with using the latest commits (it is actually a good thing that you do and report problems). However, keep in mind that changes are minimally tested and may be reverted, so if you don't have time to constantly update you code, it is better to stick to releases.
My explanation about master branch and that the change may not make it into the release was the way of saying that (and why) I am not going to announce it.

IgnasJarusevicius, 2019-07-29:
I reverted structure to its previous form in https://github.com/myriadrf/LimeSuite/commit/0b8ac78a16935c527274df19b52e7835a6a0b559, so this issue should no longer be relevant.

closed 2019-07-29 by IgnasJarusevicius